guysoft/FullPageOS #276: Xsession: unable to launch "/home/pi/scripts/start_gui"

TakuyaSan420, 2019-05-28:
Hello,
sorry for not using the template but its easier to understand like this. Basically, FullPageOS is running on my PiServer and it boots but right after the login it says " Xsession: unable to launch "/home/pi/scripts/start_gui" X session --- "/home/pi/scripts/start_gui" not found "
So my question is, where in xSession is that path specified so I can point it towards a fixed directory on piServer
Thanks

guysoft, 2019-05-29:
What version of FullPageOS?
Is this somethin you had running for a while and stopped? Or is this a new setup?

guysoft, 2019-05-30:
Also see #229 , If you can provide input it might actually push it forwards.

guysoft, 2019-06-23:
Closing if no reply

asdf1nit, 2019-07-04:
@TakuyaSan420 here is the location you want to edit.

guysoft, 2019-07-18:
No response from original poster

closed 2019-07-18 by guysoft

AlbertMN, 2022-06-18:
For anyone else experiencing this; I faced this issue when setting the default user to another username than `pi`, after the latest Raspberry Pi OS update that no longer includes the default user, I figured others might try and use other usernames as well; don't.

guysoft, 2022-06-19:
It might justify opening a new issue and fixing this.

AlbertMN, 2022-06-19:
> 

Agreed. I haven't looked much at the source, but if it's changing some static `pi` text with a variable, it would be a good solution.

I was also thinking about just comiting a note of it on the README. Not much of a fix of course.

BobRubeus, 2023-04-19:
Hi !

I tried to install the last stable version of fullpageos from the Raspberry Pi Imager,
But eac time, after every steps of the first boot I got this message : 

```
Xsession: unable to launch "/home/pi/scripts/start_gui" X session ---
"/home/pi/scripts/start_gui" not found; falling back to default session.
```

I have this message even if I modify or not the files on the boot partition.

I tried to install à classic version of Raspbian OS from Raspberry Pi Imager and everything work fine.

Any ideas ?
Thanks by advance ! :)

guysoft, 2023-11-07:
@DubbingBrothers It says there is a file missing:
"/home/pi/scripts/start_gui

Can you see if that file exists?

You can also download it from here (dont forget to chmod it to 755):
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/home/pi/scripts/start_gui

It looks like a filesystem damage. Because it should be there.

1poc, 2024-01-21:
Hi

Im quite new to this but i also get this message when i boot:

Xsession: unable to launch "/home/pi/scripts/start_gui" X session ---
"/home/pi/scripts/start_gui" not found; falling back to default session.

but when i go in to terminal i can find the file. What could be wrong?

jnunez166, 2024-03-17:
I think I have figured out what is causing this issue. When using the Raspberry PI Imager and have configured it to change the name of the default user from `pi` to `frank` the path to the start_gui is now broken. If I can see where that path is hardcoded I can alter it and test this theory out.

guysoft, 2024-03-17:
@jnunez166 You are experiencing a different issue, there is a bug if you set it to anything other than pi:
https://github.com/guysoft/FullPageOS/issues/525

HagridLeVrai, 2024-10-05:
Hi,

Got the same issue and creating a symbolic link did solve the issue.

Here's how to do it if it helps some people :
```
mkdir -p /home/pi/scripts
ln -s /home/YOURUSER/scripts/* /home/pi/scripts/
```
and give permissions to your user to read/execute what's in the directory by making him the owner of it:
```
sudo chown -R YOURUSER:YOURUSER /home/pi/scripts
```

BacherJannik, 2025-03-24:
Wie kann ich die Konfigurationen anpassen, wenn ich beim Booten den Fehler erhalte und keine Konsolenoberfläche zur Verfügung steht?

guysoft, 2025-03-24:
> Wie kann ich die Konfigurationen anpassen, wenn ich beim Booten den Fehler erhalte und keine Konsolenoberfläche zur Verfügung steht?
> Translated: How can I adjust the configurations if I receive the error during boot and no console interface is available?

You can edit the sdcard directly on a different computer.