guysoft/FullPageOS #278: Start of OS gets stuck and only says "crng init done"

dmartzol, 2019-05-31:
#### What were you doing?
Starting FullpageOS for the first time. I tried with both WIFI and Ethernet. Connected over HDMI to a screen.

#### What did you expect to happen?
Start the OS normally and either system prompt or X UI appear.

#### What happened instead?
Sistem got stuck in some step and showing only "crng init done".

#### Was there an error message displayed? What did it say?
No error message. Just "crng init done"

#### Version of FullPageOS?
Can't access /etc/fullpageos_version, but date on the image file is 2018-04-18

dmartzol, 2019-05-31:
Just read in the README it is supposed to not work with RPi 2 or later. I can't delete this issue

dmartzol, 2019-05-31:
Ok, so the Rpi is version 2 I believe. It is only the PCB that is revision 1.2. So this OS is still supposed to work with my RPi.

guysoft, 2019-06-02:
Can you try the latest RC3, non-acceleration version?
Rpi 2 should work fine.

guysoft, 2019-06-02:
Can you try the latest RC3 0.10.0, non-acceleration version?
Rpi 2 should work fine.

dmartzol, 2019-06-05:
Thanks @guysoft ! It works downloading the version you specified. The first time I went to the official mirror and downloaded the version showed in the first folder. I wasn't clear to me I had to find a different version.

closed 2019-06-05 by dmartzol