guysoft/FullPageOS #280: multiple pages/tabs dont open

gbrakcounty, 2019-06-03:
#### What were you doing?
trying to open multiple webpages specified in /boot/fullpageos.txt as http://site1.com && http://site2.com.  Also tried using "&" instead of "&&".  I tried using the FullPageDashboard but the 2nd tab either never loads or loads after a very long time.  Checked my network settings they're fine.
[Please be as specific as possible here]

#### What did you expect to happen?
multiple webpages would load
#### What happened instead?
only the first webpage loaded
#### Was there an error message displayed? What did it say?
none
#### Version of FullPageOS?
not available in the location listed below.
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2019-06-05:
That is impossible, because a webbrwoser only lets you see one page at a time. To solve this FullPageOS is shipped with an app called FullPage Dashboard as the default. You need to use ``/boot/fullpagedashboard.txt`` with newline for each new page. Also it will only use that file in the first boot.
Alternatively you can use the GUI that is part of FullPageDashboard.

closed 2019-06-05 by guysoft

gbrakcounty, 2019-06-05:
> Attention: This email came from an external source. Do not open attachments or click on links from unknown senders or unexpected emails.

It's possible. I actually have an img I created of fullpageos that opens multiple pages but I had other problems with it that's why I can't use it.

guysoft, 2019-06-05:
I don't understand what you mean by "an img you created with multiple pages". Did you use FullPageDashboard for that? Or did you generate your own image?

gbrakcounty, 2019-06-05:
Sorry I meant image but used the file extension.img