guysoft/FullPageOS #283: support for raspberry pi 4 model B

StefanFeederle, 2019-06-24:
Thanks for the awesome distro! I'm using it on 4 machines with great success.
Our internal website is quite performance hungry so I'm looking forward to upgrading to the new raspberry pi 4 model b.
Is it already supported or do you plan to support it?

guysoft, 2019-06-24:
Naturally, when I release a new version a day after it Raspberrypi foundation release a new Pi.
The image came out about two hours ago. But I hope to have a nightly out.
I have no Pi 4, so I have no way to test the dual screen option, your help trying out the nightly would he needed.

You can also follow the technical talks on OctoPi, where it seems to be more lively at the moment:
https://community.octoprint.org/t/raspberry-pi-4-is-out/10332/2

StefanFeederle, 2019-06-24:
Thanks for the quick reply. I didn't mean to rush you, take your time 😄 
Of course I'll help testing the nightly. I don't have a build setup currently, can you send me an image please?

guysoft, 2019-06-24:
There is a nightly out, but I have no way to test it

StefanFeederle, 2019-06-25:
Thanks! My raspi 4 arrives tomorrow and I'll test the nightly.

guysoft, 2019-06-26:
Any results?

StefanFeederle, 2019-06-26:
The raspi 4 arrived but I'm still waiting for my micro hdmi cable.
Currently I can only say that it responds to pings and rejects VNC connections.

StefanFeederle, 2019-06-27:
Just tested it. Screen turn black after the boot sequence is finished. This happens on either of the HDMI port. Over VNC I can see that the weppage is displayed correctly.
Do you need any debug information to troubleshoot this?

guysoft, 2019-06-27:
I need ananything you can give me, because I have no way to debug.

You can try the non-acceleration version.  Might be they changed the GPU
stuff

StefanFeederle, 2019-06-27:
I dont know much about linux debugging. Can I send you a raspi 4 to support the project?

guysoft, 2019-06-27:
You are always welcome to send anything  to support the project, at the
moment this project only goes because I am volenteering. I did not get any
donation for it so far, and frankly I don't have a running instant of it.

I have already https://piitel.co.il/ sending me a 90% discount Pi 4 as a
donation, so I will be able to test (I have n type-C or mini hdmi adapters,
so that will slow me down). However RealtimePi priority is after OctoPi and
FullPageOS which I also maintain.

asdf1nit, 2019-07-04:
When I get my pi 4's in I'll test as well.

asdf1nit, 2019-07-06:
currently building a non-accelerated version but I can confirm the above statement.

Editing the boot config and commenting out dtoverlay fixes the issue. 
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2

I'll mess around some and see if there's a combination that can be used to get 3D working.

asdf1nit, 2019-07-06:
Per one of the raspbian engineers https://www.raspberrypi.org/forums/viewtopic.php?p=1489899&sid=a650da696bb656ba5cbf32cffbcdaa31#p1489899

gpu acceleration is enabled by default on pi4 with config.txt
dtoverlay=vc4-fkms-v3d 

building fullpageos both with and without gpu acceleration added this line to my config
dtoverlay=vc4-kms-v3d

This seems to come from the gui module in CustomPiOS. While this might be needed still for older pi's it doesn't look like this needs to be done for pi 4's any longer. 

So a working pi 4 config.txt after you edit the boot file should be configured at the bottom of the file as follows.

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
#dtoverlay=vc4-kms-v3d

Where the different sections was written from on build I'm not sure, but after a build commenting out the very last line and leaving the others has the rpi 4 booting and displays's fine.

Another issue:
Out of the box only hdmi0, the one closest to the power connector, would boot. Neither the default raspbian nor FullpageOS will boot if you plug into hdmi1, you only get an initial color screen, then it all goes black and I assume kernel panics. 

To fix this on you need to run `sudo rpi-update` and update the firmware and video drivers. I haven't tried adding this to my build yet. You must do this on each image you have. After you run this and reboot it will work fine. I did have some display issues on the default raspbian desktop with resolution after applying this fix but FullPageOS worked just fine.

guysoft, 2019-07-07:
Thanks for the update, I will try it out and update once I get my Pi.
Would have to test how a new FullPageOS image should work with both new and older pis.

johnmckeever, 2019-07-09:
FWIW this worked great for us.  Now I just need to get the resolution right, as I've got a FullPageDashboard app sitting in a 1024x768 window the middle of our 4K screen.

guysoft, 2019-07-10:
@johnmckeever Please share when you how what to do, I have no hardware to test yet.
In the meantime I am trying to build a nightly, once I pass the issue with https://github.com/RPi-Distro/pi-gen/issues/309

asdf1nit, 2019-07-11:
Another issue I've found is screen blanking. I have a few kiosk with touch screens that load up a web app. 

On all previous pi's I've set the line `xset s noblank` in the `/home/pi/scripts/start_gui` to `xset s 15000 15000` so that the screen would turn off after no use for a while.  When a user needs to use it if the screen is off they tap the touch screen and it comes back on.

On the pi 4 the screen goes dark but you can still see the backlight is on. Tapping the screen does nothing and neither does any chrome keyboard shortcuts. Ctrl + w normally kills chrome and it starts back up but this doesn't work either. I did notice that if you vnc into the kiosk that you still see chrome running the page it's supposed to. This is similar to the way chrome acts on the pi before removing the line `dtoverlay=vc4-kms-v3d` from the config file so I suspect it has to do with the video driver and the screen blanking.

I haven't had enough time to troubleshoot yet.

guysoft, 2019-07-11:
@asdf1nit Thats a kernel bug, its been reported there 6 days ago:
https://github.com/raspberrypi/linux/issues/3050

johnmckeever, 2019-07-12:
> @johnmckeever Please share when you how what to do, I have no hardware to test yet.
> In the meantime I am trying to build a nightly, once I pass the issue with [RPi-Distro/pi-gen#309](https://github.com/RPi-Distro/pi-gen/issues/309)

Simply using @asdf1nit 's code at the bottom of config.txt did the trick, although we're still only getting a 1024x768 box in the middle of our 4K screen, so we've got some work to do on finding the right HDMI mode settings.
 
`[pi4]
#Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
#dtoverlay=vc4-kms-v3d`

asdf1nit, 2019-07-12:
@johnmckeever While I haven't tried a 4k screen I've had no problem with getting a full 1080 screen. I have a couple 4k screens that I'll test on next week. That being said are you using the fullpage dashboard and getting that behavior or chromium on site? have you tried it on a different site? I haven't used the fullsceen dash board as it's not a part of my build so I'm not sure about it's behavior.

@guysoft  good to know that they're working on it. In the interim I have a few tvservice scripts that I'm using with crontab to turn the tvs/monitors off at night, not sure if they would be helpful to the project, If so I can throw them into a module.

guysoft, 2019-07-14:
> @guysoft good to know that they're working on it. In the interim I have a few tvservice scripts that I'm using with crontab to turn the tvs/monitors off at night, not sure if they would be helpful to the project, If so I can throw them into a module.
I would need to seem them, depends if enough people will use them. 
If you want to maintain and you are making your own builds then a module or variant would be the best way.

asdf1nit, 2019-07-17:
> I would need to seem them, depends if enough people will use them.
> If you want to maintain and you are making your own builds then a module or variant would be the best way.

I am making my own builds so it's easy because my crontab file is always the same. Looking at this again, it would be difficult to create a script that could generate a crontab for others and work if they didn't understand the syntax of a crontab so it's better to leave this out of a module.

For anyone wanting to know what worked I'll summarize.
Create 2 files in the pi scripts dir, `tvon.sh` and `tvoff.sh`.
Contents of tvon.sh
`#!/bin/bash
tvservice -p
sleep 10
/home/pi/scripts/fullscreen`

contents of tvoff.sh
`#!/bin/bash
tvservice -p`

use `crontab -e` to create your file with the pi user and make your rules as necessary.

johnmckeever, 2019-07-17:
@asdf1nit : We got our RasPi 4 working on a 4K Sony TV without problems.  I've attached our config.txt for reference. 
[config.txt](https://github.com/guysoft/FullPageOS/files/3399655/config.txt)

guysoft, 2019-07-17:
Ok, it seems like raspberrypi 4 should be working in the nightly.
I will be able to confirm with mine soon when I get all the parts only in two weeks.

Also, we haven't tested dual screen support yet, that might take more time.

asdf1nit, 2019-07-18:
@guysoft Just got another pi in, this time the 4gb version and I can confirm that the nightly boot's and runs with only a few issues.

I flashed with etcher then I first booted into the fullpage os welcome screen all good. Trying to add a tab to the fullpage dashboard didn't work and returned google refused to connect. I find that no https sites work on the FullPageDashboard only http. I'm not sure if this was\is intended or something isn't working with it as I've never used it.

I added a different site for chromium to start at boot then rebooted and it works as intended. I tested on both a 1920x1080 TV and my 2560x1440 monitors with no issues.

Sound goes out the HDMI. 

I did try 2 screens and they both work at once but not as expected. It didn't mirror as I thought it would, though it was mirrored during boot, when chrome opened it treated them as one very large screen so that the site was split right down the middle and I had a 5120x1440 screen. I've done a little searching around but haven't seen any mention of a config.txt option. Booting into a full raspbian desktop there are a few options you can set but mirroring wasn't one of them. Side note, I did get them to mirror by accident playing with settings on the desktop but it was more of a glitch and upon reboot it was back to normal. Both screens working is possible but as what to try I'm not sure. Maybe there's a matchbox setting that needs to be set?

samhqt, 2019-07-23:
Hi,

Even I follow you and use the last version of FullPageOS it's not even booting of showing something.

Do you have similar problem ? How did you fix it ?

guysoft, 2019-07-29:
@samhqt Does normal Raspbian work for you at all?

samhqt, 2019-08-01:
@guysoft I tried to install Raspbian Buster (normal and lite) but it's always saying "Resizing filesystem" and it's reboot without doing anything else.

I don't know what can I do.

guysoft, 2019-08-01:
Ok, if raspbian isn't working then its not a FullPageOS problem.
And I would guess hardware. so check your SD card isn't faulty and you can
write to it. And that the power supply is good, and so is the power cable.

samhqt, 2019-08-01:
I expected this.

By the way do you know when the new version for FullPageOS will be out ?
Thank you for your help :)

guysoft, 2019-08-01:
When its stable.
I just released day before the Pi 4 came out,  because we were not notified.
Also the dual display is a change we might want to consider.

You can use the nightly untill we are sure no other changes should go in.

samhqt, 2019-08-03:
@guysoft I found a way to boot Raspbian.

But not for FullPageOS, do I need to add the file named "bcm2711-rpi-4-b.dtb" (found in NOOBS) to boot or not ?

Can someone gave me a list or screenshot of files they have for FullPageOS ?

samhqt, 2019-08-03:
I run FullPageOs in my RasPi 4, thank you !

When I try to display Webpage which use JS, it takes so long to display it, is it normal ?

samhqt, 2019-08-05:
I'm run FullPageOs in my RasPi 4, thank you !

When i run free -h, I see that it's not using full RAM as it should. 
I try to display Web page with JavaScript and it's not using more than 500MB, there is a way to unlimited it ?
[image]

guysoft, 2019-08-06:
@samhqt You can try and update if changing the gpu mem helps:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/start_chroot_script#L39

samhqt, 2019-08-07:
@guysoft I pushed until ggpu_mem=512 and it's start to be good but I would like to push to gpu_mem=1024 but it didn't boot until the end.
Even gpu_meme_1024=944 it's not working.

Do you know why ? Or find a way to attribute more ?

samhqt, 2019-08-22:
I could run JS application on my RasPi 4.

I found out some problems, the file libGLES2.so.2 so I installed the package libgles2-mesa and the JavaScript was running. I need it to do some GPU optimization by activate some chromium flag.

I found out Raspbian is limitating usage of memory to 1GB and you can set up on config.txt file a parameter to go around : total_mem=3072

Thank you for your help and your OS @guysoft :)

gelinger777, 2019-10-06:
Does stable release already support Pi4? Or we need to use nightly build?

guysoft, 2019-10-06:
@gelinger777 No, but the nightly does. Considering just releasing it. Since I am not finding time to get the dual screen coded.

gelinger777, 2019-10-06:
I see you are fanatically fast on answerig things. Thank you man. We need only on 1 screen to use it. So it should work out of the box, right?

guysoft, 2019-10-06:
@gelinger777 Yes. If you are using it in your business consider making a donation. I am maintaining this 100% on my spare time and it helps me make time for it. 

Latest nightly here:
http://unofficialpi.org/Distros/FullPageOS/nightly/

gelinger777, 2019-10-06:
We are working on a demo screen. Wen it goes throught, I am sure we will.

asdf1nit, 2019-10-11:
@guysoft The latest raspbian update had some fixes for dual screens. So as I have time to test I'll give an update on this

The pi 4 runs reliably on my custom builds. I don't use the dashboard module or the prebuilt images. I have had 3 pi 4's running with FullPageOS for a few months now. The few issues it does have are screen blanking and dual screens.

guysoft, 2019-10-11:
@asdf1nit I'll build a new nightly
@gelinger777 thanks for the link, didn't opened it but didn't test it on a FullPageOS instance yet

PeterDKs, 2019-10-19:
As the other have mentioned pi 4 doesn't boot at the latest official build v. 0.10.0.

Therefore I've booted at your lates nightly build v. 0.11.0 and it boots up succefully.

On the start page i can´t add any pages, there simply doesnt happen anything when i click "Add". And it's the same result on the pi's concole and on the remote administration website for the FullPageDashboard.

Here is an image where i try to add a page:
[image]

Any idea of what could be wrong?

guysoft, 2019-10-29:
@PeterDKs That just looks like the css failed to load. Try ``crtl+shift+R`` key combination to refresh teh browser.

PeterDKs, 2019-10-29:
The isser seemed to be persistent from the start. I’ll give it a shot later today.

Do you plan to release a new stable release of FullpageOS that support Raspberry pi 4?

guysoft, 2019-10-29:
@PeterDKs Yes, once I do that for OctoPi. I also moved build servers yesterday and didn't finish set up FullPageOS:
https://twitter.com/GuySoft/status/1188735297758138375

asdf1nit, 2019-11-09:
> @asdf1nit Thats a kernel bug, its been reported there 6 days ago:
> [raspberrypi/linux#3050](https://github.com/raspberrypi/linux/issues/3050)

@guysoft This has now been fixed if you update to the latest test firmware via `rpi-update` hopefully it'll get pushed to stable soon

guysoft, 2019-11-10:
@asdf1nit issue was not closed though.

Jesus080967, 2019-11-12:
Good, I think it's a great job, my English is very poor and I translate it with Google
I just received my new Raspberry Pi 4 and I am trying to rotate the screen. In previous versions of Raspberry I did it simply by setting display_rotate = 1 in config.txt, but in Raspberry 4 the screen is partially black and does not rotate correctly.

guysoft, 2019-11-13:
@Jesus080967 Its documented as not supported on the Raspberrypi 4 default graphics driver, you can try using a diffrent one:
> If using the VC4 FKMS V3D driver (this is the default on the Raspberry Pi 4), then 90 and 270 degree rotations are not supported. The Screen Configuration utility provides display rotations for this driver.

Source: https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/video.md#generic-display-options

Jesus080967, 2019-11-13:
Thanks for answering, as I could solve from config.txt file, would you have to disable the VC4 FKMS V3D driver ?, would you need to rotate the screen 90 degrees ?.
Would you have to put in front of the line (#)?

melvinisken, 2019-11-14:
Another think I came across yesterday, I hope this is the right place to mention: I was not able to activate chromium GPU support in the latest nightly image. Chromium and Raspbian seem to have received some updates / bugfixes in the last months to make it work (and I confirmed that it works using the latest Raspbian image (fully updated) using either the standard desktop environment or just x with matchbox and chromium). But within FullpageOS it didn't work (and it makes quite a difference when playing HD videos, at least with my setup).

guysoft, 2019-12-02:
Release candidate out https://github.com/guysoft/FullPageOS/issues/305

guysoft, 2020-01-09:
Released

closed 2020-01-09 by guysoft