guysoft/FullPageOS #284: Build didn't complete but boots fine.

asdf1nit, 2019-07-04:
#### What were you doing?

Building a distro with a few modifications to it. Specifically editing:
FULLPAGEOS_OVERRIDE_TIMEZONE
FULLPAGEOS_OVERRIDE_LOCALE
FULLPAGEOS_OVERRIDE_PASSWORD
FULLPAGEOS_INCLUDE_LIGHTTPD=no
FULLPAGEOS_INCLUDE_DASHBOARD=no
FULLPAGEOS_INCLUDE_WELCOME=no

All other mods were done in the FullPageOS start_chroot_script

A fix for the time zone issue that I posted here https://github.com/guysoft/FullPageOS/issues/234#issuecomment-508540758

Changing the VNC password, removing the raspicam, changing the hostname, adding another user and files to their directory.

#### What did you expect to happen?

Build to show passed in Green.

#### What happened instead?

The first build without mods ran fine. While playing around and I had other builds fail but for obvious reasons that I either fixed(tzdata) or can't implement(installing cockpit). I now have an image that works but the last of the build log is as follows

#### Was there an error message displayed? What did it say?

`Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
rootfs: 63055/179952 files (0.1% non-contiguous), 447480/724992 blocks
++++ tune2fs -l /dev/loop0
++++ grep -i 'block size'
++++ awk -F: '{print $2-0}'
+++ e2fblocksize=4096
++++ resize2fs -P /dev/loop0
++++ grep -i 'minimum size'
++++ awk -F: '{print $2-0}'
+++ e2fminsize=580403
+++ e2fminsize_bytes=2377330688
+++ e2ftarget_bytes=2587045888
+++ e2fsize_bytes=2969566720
+++ e2fminsize_mb=2267
+++ e2fminsize_blocks=4643225
+++ e2ftarget_mb=2467
+++ e2ftarget_blocks=5052825
+++ e2fsize_mb=2831
+++ size_offset_mb=364
+++ losetup -d /dev/loop0
+++ echo 'Actual size is 2831 MB (5799936 blocks), Minimum size is 2267 MB (580403 file system blocks, 4643225 blocks)'
Actual size is 2831 MB (5799936 blocks), Minimum size is 2267 MB (580403 file system blocks, 4643225 blocks)
+++ echo 'Resizing to 2467 MB (5052825 blocks)'
Resizing to 2467 MB (5052825 blocks)
+++ '[' 364 -gt 0 ']'
+++ echo 'Partition size is bigger then the desired size, shrinking'
Partition size is bigger then the desired size, shrinking
+++ shrink_ext 2019-06-20-raspbian-buster-lite.img 2 2466
+++ image=2019-06-20-raspbian-buster-lite.img
+++ partition=2
+++ size=2466
+++ echo 'Resizing file system to 2466 MB...'
Resizing file system to 2466 MB...
++++ awk '{print $4-0}'
++++ grep 2019-06-20-raspbian-buster-lite.img2
++++ sfdisk -d 2019-06-20-raspbian-buster-lite.img
+++ start=540672
+++ offset=276824064
+++ detach_all_loopback 2019-06-20-raspbian-buster-lite.img
++++ losetup
++++ grep 2019-06-20-raspbian-buster-lite.img
++++ awk '{ print $1 }'
+++ for img in $(losetup  | grep $1 | awk '{ print $1 }' )
+++ losetup -d /dev/loop0
losetup: /dev/loop0: detach failed: No such device or address
+ exit 1`

#### Version of FullPageOS?
0.11.0

guysoft, 2019-07-07:
1. What version of CustomPiOS are you using?
2. What build method are you using? Docker/Vagrant?

asdf1nit, 2019-07-07:
Just cloned both CustomPiOs and FullPageOS to a local debian box and built locally, no Docker or Vagrant.

`sudo apt-get install realpath p7zip-full qemu-user-static

git clone https://github.com/guysoft/CustomPiOS.git
git clone https://github.com/guysoft/FullPageOS.git
cd FullPageOS/src/image
wget -c --trust-server-names 'https://downloads.raspberrypi.org/raspbian_lite_latest'
cd ..
../../CustomPiOS/src/update-custompios-paths
sudo modprobe loop
sudo bash -x ./build_dist`

guysoft, 2019-07-07:
> raspberrypi.org/raspbian_lite_latest'

Ok, what distribution are you using?
It looks like your distro has different behavior of the loopback mount.

What is the output of losetup?

asdf1nit, 2019-07-08:
I'm using Debian, So I'm new to building like this, but losetup returns nothing, on one shell and command not found on another. Terminal for the former and an ssh session from my main computer on the other.

guysoft, 2019-07-08:
What version of Debian?

It makes no sense that you are getting no return on one command, and "command not found in the other. I can't support an inconsistent environment. Its something I can't reproduce.

I urge you to use either Vagrant or Docker build solution. Since its there to keep your environment consistent. Otherwise its very unlikely we will be able understand what is not working.

asdf1nit, 2019-07-08:
I will switch to docker.

asdf1nit, 2019-07-09:
Never used docker for this.

Is this expected?

custompios-watcher | time="2019-07-09T00:09:14Z" level=info msg="Unable to update container /ArkpiKiosk-build, err='/config.json - read /config.json: is a directory'. Proceeding to next." 
custompios-watcher | time="2019-07-09T00:09:44Z" level=error msg="Unable to find default config file /config.json - read /config.json: is a directory" 
custompios-watcher | time="2019-07-09T00:09:44Z" level=info msg="Unable to update container /ArkpiKiosk-build, err='/config.json - read /config.json: is a directory'. Proceeding to next." 
custompios-watcher | time="2019-07-09T00:10:14Z" level=error msg="Unable to find default config file /config.json - read /config.json: is a directory" 
custompios-watcher | time="2019-07-09T00:10:14Z" level=info msg="Unable to update container /ArkpiKiosk-build, err='/config.json - read /config.json: is a directory'. Proceeding to next." 
custompios-watcher | time="2019-07-09T00:10:44Z" level=error msg="Unable to find default config file /config.json - read /config.json: is a directory" 
custompios-watcher | time="2019-07-09T00:10:44Z" level=info msg="Unable to update container /ArkpiKiosk-build, err='/config.json - read /config.json: is a directory'. Proceeding to next." 
custompios-watcher | time="2019-07-09T00:11:14Z" level=error msg="Unable to find default config file /config.json - read /config.json: is a directory" 
custompios-watcher | time="2019-07-09T00:11:14Z" level=info msg="Unable to update container /ArkpiKiosk-build, err='/config.json - read /config.json: is a directory'. Proceeding to next." 
custompios-watcher | time="2019-07-09T00:11:44Z" level=error msg="Unable to find default config file /config.json - read /config.json: is a directory" 
custompios-watcher | time="2019-07-09T00:11:44Z" level=info msg="Unable to update container /ArkpiKiosk-build, err='/config.json - read /config.json: is a directory'. Proceeding to next."

asdf1nit, 2019-07-10:
> What version of Debian?
> 
> It makes no sense that you are getting no return on one command, and "command not found in the other. I can't support an inconsistent environment. Its something I can't reproduce.
> 
losetup is now built into the kernel of debian so the command `sudo modprobe loop` no longer functions in the intended way.

guysoft, 2019-07-10:
Indeed, not sure how it behaves in most environments now, so will have to figure out how this effects build instructions.

asdf1nit, 2019-07-11:
For my environment not using the command works just fine in most cases but not all and when it doesn't the build is still fine just not shrunk down completely. My original comment show's what happens when the check is run but it's not consistent, it completes in most cases just not all.

closed 2019-07-11 by asdf1nit

guysoft, 2019-07-11:
@asdf1nit It sounds like its not releasing the loopback device, and when its running the fsck scan it becomes inconsistent. I added a few fixes to that which you can make sure you are using:
This:
https://github.com/guysoft/CustomPiOS/commit/e9d972eeddaf1b7c1d642e73a4f97c5ec2e6143b

And this:
https://github.com/guysoft/CustomPiOS/commit/7c48f209e20ad9ef04c7780860158b3a69dc2915

Related issue:
https://github.com/guysoft/CustomPiOS/issues/46