guysoft/FullPageOS #288: chromium takes long to load

gbrakcounty, 2019-07-19:
I have a pi set up with fullpageos in my lab which works fine.  I created an IMG of the SD card, then imaged another blank SD card, put that into a new Pi, booted it up, and chromium doesn't load until about 3-5min later.  i also noticed that the vnc server doesn't load until chromium loads as well so I'm wondering if the desktop environment is being delayed somehow in the ~/scripts folder?  Again, no issues whatsoever in my lab environment with the same config.

asdf1nit, 2019-07-30:
@gbrakcounty only time I've ever had that issue is when the pi is connected to wifi. When I'm on a wired connection it's always fast. Try that an see is if makes a difference.

Also there's no need to edit the `~/scripts/start_chromium_browser.` file to point to a site. You edit `/boot/fullpageos.txt` which just contains the site.

gbrakcounty, 2019-07-30:
> Attention: This email came from an external source. Do not open attachments or click on links from unknown senders or unexpected emails.

I'm using Ethernet. Also, this issue only seems to occur when connected to a tv via HDMI.

I use the start_chromium_browser file to open multiple pages. I use an extension to show tabs.

guysoft, 2019-07-31:
Ok, if you try a different HDMI screen does it take less?
What version of RaspberryPi are you using?

gbrakcounty, 2019-07-31:
It works fine when using an HDMI to DVI cable connected to a computer monitor. I don't have a other tv to test at the moment. I'm using version 3 B+.

guysoft, 2019-08-20:
If its dependent on the HDMI hardware then I am not sure how I can debug it, unless there is something else to go on. Have you tried regular rapbian?

guysoft, 2020-06-27:
Closing if no reply