guysoft/FullPageOS #289: unable to rotate hdmi display

gbrakcounty, 2019-07-19:
Not sure if this is specific to fullpageos or not, but since its a custom os and something standard to raspbian isnt working, i figured id post here.  display_rotate and display_hdmi_rotate do not work at all in /boot/config.txt.  I am trying to use it for a digital ticket counter display and the client wants the tv in portrait mode.  the tv is old and does not have a rotate function.  any suggestions?

closed 2019-07-19 by gbrakcounty

guysoft, 2019-08-05:
How did you solve your issue?

gbrakcounty, 2019-08-05:
> Attention: This email came from an external source. Do not open attachments or click on links from unknown senders or unexpected emails.

“display_hdmi_rotate=3” in /boot/config.txt works only if the GL driver is disabled in raspi-config.

guysoft, 2019-08-06:
Yes, that is not implemented by the driver, 
Related to: https://github.com/guysoft/FullPageOS/issues/137

thfrei, 2022-05-30:
I updated the FAQ: https://github.com/guysoft/FullPageOS/wiki/FAQ#how-to-rotate-the-screen-to-a-vertical-position, please also see: #421