guysoft/FullPageOS #290: Black screen on pi3 model À+

Glujaz, 2019-08-01:
#### What were you doing?

Booting the OS after creating the SD card. At first boot it was ok. I had a settings page somehow. After that, nothing. There is no cursor or anything appearing, and the model A has no Ethernet, so no SSH.

#### What did you expect to happen?

See at least the full page dashboard appear.

#### What happened instead?

Black screen appears, but nothing else, and nothing is happening on the SD card
#### Was there an error message displayed? What did it say?
Nothing
#### Version of FullPageOS?

The last normal one (as of 1 August). Can't get access to the file as no keyboard or ssh available...

Thanks in advance for your help.

Glujaz, 2019-08-01:
Ok, was my monitor. I use an Apple Cinema Display HD, which is very picky with input type... It works fine with raspian, but not with FullPageOS though...

Now is connecting to WiFi... and while I set everything right, it's not working...

closed 2019-08-01 by Glujaz

guysoft, 2019-08-01:
Did you solve your issue in the end?

Glujaz, 2019-08-01:
Yes I did. Though, the raspi 3 A+ is slow...
Sorry to say, but after tweaking a little bit with FullpageOS, not having auto refresh is very limiting...

Sorry for the opened issue.