guysoft/FullPageOS #292: clean chromium cache, cookies, passwords with script

moma73, 2019-08-05:
#### What were you doing?
I tried to clean chromium cache, cookies, passwords with a script that includes bleachbit.
#!/bin/bash
killall chromium-browser
bleachbit -c chromium.cache
bleachbit -c chromium.cookies
bleachbit -c chromium.passwords

#### What did you expect to happen?
clean chromium cache, cookies, passwords with script

#### What happened instead?
It didn't work because chromium was still running. Is there a possibility to stop chromium completely to clean cache and so on ?

#### Was there an error message displayed? What did it say?
Erreurs : 1
debug: process 'chromium-browser' is running

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2019-08-05:
You can stop the graphical interface and then run it
``/etc/init.d/lightdm stop``
[script]
``/etc/init.d/lightdm start``

moma73, 2019-08-07:
Thanks for your quick reply!
I did as you said, but now the kiosk page isn't running any more.  As I don't had the time to investigate what happened, I installed the OS from scratch. 
I'll look out for a solution later, I'll try not to use bleachbit.

guysoft, 2019-08-07:
That stops the kios so you can run stuff such as bleachbit

moma73, 2019-08-09:
Yes, I got that, it's just that after running bleachbit and restarting the kiosk, it didn't work anymore. The kiosk page did'nt load. I'll try another method to clear the cache when I have some time. If anybody has found a good method for clearing cache and passwords, I'm all ears. It's a public kiosk and also I deactivated the password saving via the Chromium menu, I want to make sur to clear passwords, forms and cache regularly.

asdf1nit, 2019-08-13:
Is your killall command working? That command doesn't on my kiosk. Try using ` sudo pkill chromium` then clearing the cookies and things you want. Chrome will automatically start back up right after you issue the command.

Also I see your using bleachbit you don't have to use that you can just clear that with a simple script.
https://askubuntu.com/questions/391858/clearing-all-private-user-data-from-chrome-from-the-command-line

moma73, 2019-08-23:
Hey, thanks a lot for your help !
I finally got what I wanted. That's the script I use :+1: 

#!/bin/bash
sudo /etc/init.d/lightdm stop
rm -rf /home/pi/.cache/chromium/Default
rm -rf /home/pi/.config/chromium/Default/Login\ Data
rm -rf /home/pi/.config/chromium/Default/Cookies
rm -rf /home/pi/.config/chromium/Default/History
rm -rf /home/pi/.config/chromium/Default/Web\ Data
sudo reboot

When I use "sudo pkill chromium", it seems that the process is not killed long enough to execute the script. But "sudo /etc/init.d/lightdm stop" does do the job. I had to delete the "sudo pkill chromium", it made the script stop at the second line, I guess. 
Also, I couldn't make the kiosk work properly with "/etc/init.d/lightdm start" at the end of the script, so I added the reboot. I wanted to reboot it anyway once or twice à day. 

You might close the issue, I let it open in case you want to add a line.

asdf1nit, 2019-08-24:
@moma73 glad you found what you needed. 

Just an observation

>#!/bin/bash
>sudo /etc/init.d/lightdm stop
>rm -rf /home/pi/.cache/chromium/Default
>rm -rf /home/pi/.config/chromium/Default/Login\ Data
>rm -rf /home/pi/.config/chromium/Default/Cookies
>rm -rf /home/pi/.config/chromium/Default/History
>rm -rf /home/pi/.config/chromium/Default/Web\ Data
>sudo reboot

I don't think you need all of those lines. Since `rm -rf` is recursive, anything under `/home/pi/.cache/chromium/Default` is removed after the 4th line making 5,6 7 of no purpose.

I'm going to keep this in mind and roll it into the admin-toolkit module in the future. I haven't had the need for this yet but I see it's usefulness.

moma73, 2019-08-26:
Thanks for your reply.

I don't think that anything under /home/pi/.config/chromium/Default is removed, and it's not what I wanted to do, too. There might be configuration data in this Default folder, that I need to keep.

I just picked the "Login Data", "Cookies", "History" and "Web Data" files to remove them, I think it's not necessary to do it recursevily, in fact. And there is much more data that can be deleted in this Default folder, but I don't know what it is exactly.

The line "rm -rf /home/pi/.cache/chromium/Default" has to be recursive to delete everything in that folder.
See, what I mean??

t-knapp, 2023-03-29:
> Hey, thanks a lot for your help ! I finally got what I wanted. That's the script I use 👍
> 
> #!/bin/bash sudo /etc/init.d/lightdm stop rm -rf /home/pi/.cache/chromium/Default rm -rf /home/pi/.config/chromium/Default/Login\ Data rm -rf /home/pi/.config/chromium/Default/Cookies rm -rf /home/pi/.config/chromium/Default/History rm -rf /home/pi/.config/chromium/Default/Web\ Data sudo reboot
> 
> When I use "sudo pkill chromium", it seems that the process is not killed long enough to execute the script. But "sudo /etc/init.d/lightdm stop" does do the job. I had to delete the "sudo pkill chromium", it made the script stop at the second line, I guess. Also, I couldn't make the kiosk work properly with "/etc/init.d/lightdm start" at the end of the script, so I added the reboot. I wanted to reboot it anyway once or twice à day.
> 
> You might close the issue, I let it open in case you want to add a line.

This one worked for me.