guysoft/FullPageOS #293: Turn monitor on again

wolfio, 2019-08-06:
#### What were you doing?
I'm turning off my monitor with the help of mqtt which executes
`tvservice -o`
This works fine.

With version 0.7.0 I just executed
`tvservice -p && chvt 1 && chvt 7`
and it turned on again

I upgraded now to version 0.10.0 and since then this is not working anymore.

I also tried some other solutions I found without success, e.g.
`tvservice -p
sleep 10
/home/pi/scripts/fullscreen`

Is there a way to turn on the monitor again without a reboot.

#### What did you expect to happen?
That the monitor is turned on again.

#### What happened instead?
nothing

#### Was there an error message displayed? What did it say?
If I execute tvservice -p :
`Powering on HDMI with preferred settings
[E] Failed to power on HDMI with preferred settings`

#### Version of FullPageOS?
previously 0.7.0 - now 0.10.0

guysoft, 2019-08-06:
Seems like raspbian changed something. How about trying this?
https://raspberrypi.stackexchange.com/a/78330/7483

wolfio, 2019-08-06:
Tried that yesterday already without success. Did try it now again and -> it works. Thanks very much. Probably I mixed it yesterday by still turning the monitor off with tvservice.

closed 2019-08-06 by wolfio