guysoft/FullPageOS #295: Official Mirror Not Booting on 3B+

conrad9900, 2019-09-04:
#### What were you doing?

Trying to boot on 3B+

#### What did you expect to happen?

Fullpageos landing page

#### What happened instead?

It runs through file expansion, reboots, shows scrolling information with raspberry pi logos on top, once finished it goes black with no change after 10 minutes. I have tried to:

- Re download the image
- Re-write the image 4 times to SD
- Changes the SD card
- Changed the PI

#### Was there an error message displayed? What did it say?
Just a black screen

#### Version of FullPageOS?

2019-06-17_2019-04-08-fullpageos-stretch-lite-0.10.0.zip
--

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2019-09-04:
Hey, you already opened a similar issue at: https://github.com/guysoft/FullPageOS/issues/235

Have you solved the issue then? You keep posting that the image does not work for you but you don't report if the issue has been resolved.

You also didn't respond to this issue which you posted:
https://github.com/guysoft/FullPageOS/issues/230

conrad9900, 2019-09-04:
Hi, in regards to #235, I had to move back to 2018-03-13-fullpageos-stretch to resolve the black screen issue, however that images touchscreen controls do not calibrate correctly as mentioned in #230, the cursor is 2-3cm away from the touch point, even after re calibration. I was using 3B pi,s until now as they are not affected by the touchscreen issue. I have moved back to the official build now to see if it has fixed the issue as i am upgrading to 3B+ only, however all i get is a black screen. apologies for the non reply as i had disabled notification from ghub due to spam.

conrad9900, 2019-09-05:
Got the issue resolved.
The pi would not show on 7" touchscreen, only black screen.
Attaching the pi to a monitor via hdmi and disabling overscan and acceleration fixed the issue

closed 2019-09-05 by conrad9900