guysoft/FullPageOS #296: Unable to build FullPageOs on a fresh install with 2019-04-08-raspbian-stretch-lite

jarrah31, 2019-09-06:
#### What were you doing?
Manually building FullPageOS because I also need to install Hyperpixel (https://shop.pimoroni.com/products/hyperpixel-4)

I originally tried using 2019-06-17_2019-04-08-fullpageos-stretch-lite-0.10.0.zip but the hyper pixel software wouldn't install on top of that.

I've also tried Buster but that fails in a different place. Will post details when I try Buster again.

I've previously had this working really well when I built one a year ago, but the SD card failed so I'm trying to rebuild and remember which order I installed it all with.

#### What did you expect to happen?
"sudo bash -x ./build_dist" to work

#### What happened instead?
It failed with the following error:
```
 Unmounting /home/pi/FullPageOS/src/workspace/mount...
++++ sudo umount /home/pi/FullPageOS/src/workspace/mount
++++ echo_red -e '\nBUILD FAILED!\n'
++++ echo -e -n '\e[91m'
++++ echo -e '\nBUILD' 'FAILED!\n'

BUILD FAILED!

++++ echo -e -n '\e[0m'
+ exit 1
```

#### Version of FullPageOS?
v0.10

#### Screenshot(s) showing the problem:
Logs above

Given that the following command downloads Buster:
`wget -c --trust-server-names 'https://downloads.raspberrypi.org/raspbian_lite_latest'`

I downloaded Stretch using the following instead. This was the only change I made to the build instructions.
`wget -c --trust-server-names 'https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/2019-04-08-raspbian-stretch-lite.zip'`

Full output of install here:
https://pastebin.com/ZKUVbjUt

guysoft, 2019-09-08:
Hey,
Ok the error is at:
```
+ apt-get update --allow-releaseinfo-change
E: Command line option --allow-releaseinfo-change is not understood in combination with the other options
```
The errors you posted  that appear after it are what happens after that command fails (namely clenaup).

The reason this the option `` --allow-releaseinfo-change`` is there because there were issues the apt-get in debian when buster came out.

As a workaround you can remove from ``--disable-services`` module in CustomPiOS, and every other place it appears. 

I need to see if removing that flag does not break anything else, and if so this fix can be added to all of CustomPiOS.

jarrah31, 2019-09-08:
Thanks for looking into this Guy! 
Buster is the way forward so let me try that again and I'll report back on the error it gave.

jarrah31, 2019-09-08:
ok, here is the error a fresh install of Buster Lite on a Raspberry Pi gives when attempting to manually install FullPageOS:
```
Resizing file system to 2503 MB...
++++ sfdisk -d 2019-07-10-raspbian-buster-lite.img
++++ grep 2019-07-10-raspbian-buster-lite.img2
++++ awk '{print $4-0}'
+++ start=540672
+++ offset=276824064
+++ detach_all_loopback 2019-07-10-raspbian-buster-lite.img
++++ losetup
++++ grep 2019-07-10-raspbian-buster-lite.img
++++ awk '{ print $1 }'
+++ for img in $(losetup  | grep $1 | awk '{ print $1 }' )
+++ losetup -d /dev/loop0
losetup: /dev/loop0: detach failed: No such device or address
+ exit 1
```
Full output here: https://pastebin.com/VzugWcqE

Thank you!

guysoft, 2019-09-08:
1. Are you trying to install FullPageOS or build an image?
You said initially you want to build it, but not you used the word "install"
2.

The output is strange, it says initially that you have /dev/loop0 but then it fails to detach.

What is the output of ``loseup``?
It might be that you ran the command serval times and you have some loopback mount that got stuck, a reboot would fix it for sure if that is the case.

jarrah31, 2019-09-09:
ah I see, I may have misunderstood the end result of the instructions titled "Build FullPageOS From within FullPageOS / Raspbian / Debian / Ubuntu". What I thought I was doing is manually "installing" FullPageOS, but now I understand that build relates to building an image. 

Is there a way to manually install FullPageOS please?

guysoft, 2019-09-09:
@jarrah31 No there isn't.
There might be a growing demand to add to CustomPiOS an option that lets it run directly on an image, thus enabling this. However I have no time to add it. Furthermore it would potentially go against the ideology of the CustomPiOS which is: "Once you build it, you can deploy it evreywhere and get the exact same thing. 

Do hyperpixel-4 provide an image you can flash? You should be able to build FullPageOS on top of that.

jarrah31, 2019-09-09:
Thanks for the explanation, I'm happy with the way you have things set up. I will write the FullPageOS image to an SD card and just work on installing Hyperpixel software on there. 
Thank you for your help!

closed 2019-09-09 by jarrah31