guysoft/FullPageOS #297: Chromium Remembers Password but does not auto-login

conrad9900, 2019-10-22:
#### What were you doing?

Setting a username and password field, saving it in chromium when prompted.

#### What did you expect to happen?
When rebooted, the homepage should auto login with saved credentials.
I am forced to delete all history and repeat the process 3-4 times before it will auto-login.

#### What happened instead?
Login screen is shown with the username and password field autofilled

#### Was there an error message displayed? What did it say?
No error message

#### Version of FullPageOS?
2018-03-13-fullpageos-stretch 0.9.0

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2019-10-29:
What website?
Auto-login works fine for me for sevral websites.

guysoft, 2021-08-31:
Closing due to no reply

closed 2021-08-31 by guysoft