guysoft/FullPageOS #298: More complete docker instructions

semanticfire, 2019-10-24:
#### What were you doing?

Building the latest FullPageOS
checked out this repo, moved in the src directory and followed the instructions on the link:
https://github.com/guysoft/CustomPiOS/wiki/Building-with-Docker

#### What did you expect to happen?
A minimal working image

#### What happened instead?
Nothing, docker starts, creates image and exits no errors

#### Was there an error message displayed? What did it say?
none

#### Version of FullPageOS?

latest from GIT

guysoft, 2019-10-29:
1. Are you using ``docker run`` or docker-compose?

You need to use the container to run a build. As shown in the readme:
```
docker exec -it mydistro_builder /CustomPiOS/nightly_build_scripts/custompios_nightly_build
```

semanticfire, 2019-10-29:
I use the `docker run`, but the container is not starting, so the exec fails

guysoft, 2019-10-29:
Try running it like this:
```
docker run -t -v `pwd`:/distro --name mydistro_builder --device /dev/loop-control guysoft/custompios:devel  /CustomPiOS/nightly_build_scripts/custompios_nightly_build
```

I usually use the docker compose to build multiple times on the same container.

Note you might need to run:
```
sudo docker rm -f mydistro_builder
```
Because you already created that container.

semanticfire, 2019-10-29:
This generates way more output:
https://gist.github.com/semanticfire/daad7560de9f173c34503901a974974b

guysoft, 2019-10-29:
That command should rename the file correctly,  not sure why it doesn't in
your case.

Did you get an image? Can you post your build.log?

semanticfire, 2019-10-29:
Build logs attached for building a variant, I haven't tested the image yet but at least it builds one.
I delete my existing container and start the build with the comment above ( image renamed manually )
The build fails at one point see log:
[build-1.log](https://github.com/guysoft/FullPageOS/files/3785257/build-1.log)
 
Then I delete the container and restart the build and then it works.
[build.log](https://github.com/guysoft/FullPageOS/files/3785258/build.log)

guysoft, 2019-10-30:
Looks like it worked:

```
BUILD SUCCEEDED!
``` 
In green means it worked.

Note you are not building a variant, to do that you need to sesify which variant.

I am not sure what you are doing, it looks like the variant has the a copy of the fullpageOS source.

A variant is code that runs before and after, should not be a complete copy.
If you want to fork FullPageOS you should edit that module. But note it will not be easy to maintain against mainline FullPageOS.

Saying because I saw this:
```
+++ '[' -d /distro/variants/ri-bloom/filesystem ']'
+++ cp -vr --preserve=mode,timestamps /distro/variants/ri-bloom/filesystem .
'/distro/variants/ri-bloom/filesystem' -> './filesystem'
'/distro/variants/ri-bloom/filesystem/boot' -> './filesystem/boot'
'/distro/variants/ri-bloom/filesystem/boot/fullpageos.txt' -> './filesystem/boot/fullpageos.txt'
'/distro/variants/ri-bloom/filesystem/boot/fullpageos.txt~' -> './filesystem/boot/fullpageos.txt~'
'/distro/variants/ri-bloom/filesystem/etc' -> './filesystem/etc'
'/distro/variants/ri-bloom/filesystem/etc/default' -> './filesystem/etc/default'
'/distro/variants/ri-bloom/filesystem/etc/default/keyboard' -> './filesystem/etc/default/keyboard'
'/distro/variants/ri-bloom/filesystem/etc/wpa_supplicant' -> './filesystem/etc/wpa_supplicant'
'/distro/variants/ri-bloom/filesystem/etc/wpa_supplicant/wpa_supplicant.conf' -> './filesystem/etc/wpa_supplicant/wpa_supplicant.conf'
'/distro/variants/ri-bloom/filesystem/home' -> './filesystem/home'
'/distro/variants/ri-bloom/filesystem/home/pi' -> './filesystem/home/pi'
'/distro/variants/ri-bloom/filesystem/home/pi/scripts' -> './filesystem/home/pi/scripts'
'/distro/variants/ri-bloom/filesystem/home/pi/scripts/run_onepageos' -> './filesystem/home/pi/scripts/run_onepageos'
'/distro/variants/ri-bloom/filesystem/home/pi/scripts/run_onepageos~' -> './filesystem/home/pi/scripts/run_onepageos~
``

guysoft, 2019-10-31:
Ok,
I can you give me clear reproduction commands?
What set of commands should I run to get a failure?
Can you write exactly the sequence of commands you are doing?

This works for me:

```
docker run --privileged -t -v $(pwd):/distro --name mydistro_builder --device /dev/loop-control guysoft/custompios:devel /CustomPiOS/nightly_build_scripts/custompios_nightly_build
sudo docker rm -f mydistro_builder
docker run --privileged -t -v $(pwd):/distro --name mydistro_builder --device /dev/loop-control guysoft/custompios:devel /CustomPiOS/nightly_build_scripts/custompios_nightly_build
```

semanticfire, 2019-10-31:
I don't run docker as sudo but apart from that I do the same, the first try fails:
https://github.com/guysoft/FullPageOS/files/3785257/build-1.log
Then the 2nd try works
https://github.com/guysoft/FullPageOS/files/3785258/build.log

guysoft, 2020-02-29:
@semanticfire Hey sorry for the late reply, that is a known issue. see:
https://github.com/guysoft/CustomPiOS/issues/55

Which comes from:
https://github.com/moby/moby/issues/27886

semanticfire, 2020-03-07:
@guysoft let me try this next week

semanticfire, 2020-03-09:
@guysoft running 'losetup -f' before the docker command seems to work

semanticfire, 2020-03-09:
> @guysoft running 'losetup -f' before the docker command seems to work

too soon, still fails now and then... where the 2nd try works

guysoft, 2023-10-26:
Can I close this?

closed 2023-10-26 by semanticfire