guysoft/FullPageOS #300: found an exploit while running chromium is kiosk mode

gbrakcounty, 2019-11-01:
I use fullpageos with chromium is kiosk mode and I managed to lock down chromium real tight.  disabled most keyboard shortcuts and prevented new windows/tabs from being opened which would expose the settings button.  but now after testing another internal site that we have, it seems to open a new window, well sort of.  It is indeed a new window, but smaller and not really a full window.  it is one of those mini-windows that does not any navigation buttons and has a button at the bottom of it that says "close window".  I was told this is considered a popup UI.

anyway, it exposes the address bar, which has the icon to the left of it that once clicked it shows chromium's info about how the site is being handled.  "Connection is secure" then below that there are three buttons:
Certificate
Cookies
Site settings

Site settings is the exploit.  Once i click that i gain access to chrome://settings, which can also lead to other menus such as extensions.  I am looking for an extension that forcibly hides the address bar in all windows but not having any luck.  any suggestions?

gbrakcounty, 2019-11-01:
sorry i just fixed it.  had to put --app= before the desired URL.

gbrakcounty, 2019-11-06:
ok so i found another hole.  when bringing up the print dialog window, the "Change" button under the destination printer, brings up another window that has two links that open a new windows regardless of how strict the popup blocker is.  One is the "Manage" link that opens chrome://settings/printers and the other is at the bottom of the screen, where Google offers the option to "Sign in" to print from Google Docs.  The "Sign in" link opens another windows, again ignoring popup blockers, which asks to sign into Google.  Not sure if this is an issue with fullpageos or just chromium.  but it looks like the --kiosk switch doesnt fully lock down chromium.

guysoft, 2019-11-06:
This is a chromium bug you will have to report to Chromium. I don't recompile it or anything, just ship what is on the OS
https://bugs.chromium.org/p/chromium/issues/list

closed 2019-11-06 by guysoft