guysoft/FullPageOS #302: Dual screen shows in Extended mode all time

McLund, 2019-11-14:
Hi.

I have a new Raspberry Pi4B updated to last firmware (rpi-update).

To this Pi4 is two new HDMI Philips monitors connected.

The /boot/config.txt file is unchanged.

The problem is that at the screen shows in Extended mode all time. I want to show same screen on both monitors in duplicate mode.

OBS. If i run Raspberry´s original raspbian-buster-full desktopversion the screen work´s perfect in duplicate mode.

Is there maybe anyone who can help me to setup an working /boot/config.txt file?

Thank´s for all help!   ;-)

Best regard´s,
McLund

[image]

chriswong11, 2019-11-22:
Try this:
sudo nano /etc/lightdm/lightdm.conf

Under [Seat:*]
Add: display-setup-script=xrandr --output HDMI-2 --same-as HDMI-1

[image]

Hope this helps for now.

McLund, 2019-11-25:
Hi Chris.

Thank you for your help.

I have now tested your solution but it will not work.

The two screens shows still in Extended mode as before.

Best regard´s
McLund

[IMG_20191122_111016]<https://user-images.githubusercontent.com/10913196/69441880-7bc3e280-0d19-11ea-8938-6cfd654aaf07.jpg>

guysoft, 2019-11-25:
@chriswong11 Should we add that to the distro? I'd like two screens, but having a mirror is a good start.

asdf1nit, 2020-02-10:
@guysoft  This fits my needs perfectly. I have some some plans to add 2 signage displays back to back and wanted a mirrored setup. If this could be added as a Config option in fullpageOS it would be great. If you would rather go with a module option I'm thinking about adding it to the admin toolkit as an option, along with screen blanking/display options.

Vetal7979, 2024-04-02:
> Try this: sudo nano /etc/lightdm/lightdm.conf
> 
> Under [Seat:*] Add: display-setup-script=xrandr --output HDMI-2 --same-as HDMI-1
> 
> [image]
> 
> Hope this helps for now.

I have HDMI monitor and LCD DPI screen 7" ( https://www.waveshare.com/wiki/7inch_LCD_for_Pi )
can it works in duplicate mode?
(RPI 4B, FullPageOS stable ver)

Vetal7979, 2024-04-03:
xrandr command out this:
Screen 0: minimum 320 x 200, current 1600 x 900, maximum 7680 x 7680
HDMI-1 connected primary 1600x900+0+0 (normal left inverted right x axis y axis) 443mm x 249mm
   1600x900      60.00*+
   1440x900      74.98    59.89  
   1280x800      74.93    59.81  
   1152x864      75.00  
   1024x768      75.03    70.07    60.00  
   832x624       74.55  
   800x600       72.19    75.00    60.32    56.25  
   640x480       75.00    72.81    66.67    59.94  
   720x400       70.08  
HDMI-2 disconnected (normal left inverted right x axis y axis)
DPI-1 connected 1024x600+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   1024x600      60.00*+

in ./pi/home/scripts/start_gui.sh  before string  /home/scripts//run_onepageos i add:
xrandr --output DPI-1 --mode 1024x600 --same-as HDMI-1

screen begin work in duplicate mode, but resolution on DPI-1 is not 1024x600

ndbroadbent, 2024-06-01:
I'm using the nightly release of FullPageOS on a Raspberry Pi 4B, and I'm trying to figure out how to get extended display working on two monitors. I can see the logs and splash screen on both screens when the pi boots, but the second screen goes blank once chromium starts. So far I've tried adding `xrandr --output HDMI-2 --same-as HDMI-1` to `~/scripts/start_gui`, and I've also tried adding `display-setup-script=xrandr --output HDMI-2 --same-as HDMI-1` to the `[Seat:*]` section in `/etc/lightdm/lightdm.conf`. Neither of those seem to make any difference.

The only other change I've made to my setup is setting `DISPLAY_ORIENTATION=right` in `~/scripts/start_gui` for my vertical monitor.

guysoft, 2024-06-30:
Might be worth adding that to the wiki