guysoft/FullPageOS #303: switch off tracking URL

KronnNZ, 2019-11-14:
where about is this added into the new build?  is it possible to turn this tracking data off?

fullpageos runs on a network firewalled from the internet and this URL keeps getting blocked.

https://fullpageos-tracking.gnethomelinux.com

line added in 0.11
  export USAGE_STATISTICS_URL=https://fullpageos-tracking.gnethomelinux.com

cheers,

guysoft, 2019-11-23:
Hey,
Its quite clear in the code where to disable it.
Firsr, you are clearly using FullPageOS for your professional work.
I am asking you to please consider making a donation.
I devleop FullPageOS 100% out of my free time and its source is open.

You will have an answer, but before getting it from me please try and understand how the code works yourself, it is my interest that more people do and have the tools and knowledge to contribute back.

guysoft, 2019-11-28:
Closing if no reply

closed 2020-01-09 by guysoft

My-Random-Thoughts, 2022-02-24:
sudo systemctl stop usage-statistics
sudo systemctl disable usage-statistics
sudo rm /usr/bin/boot_report

guysoft, 2022-02-24:
Alas, I have been documented by others.
Will note that I only keep the telemetry to know what to develop more, and if version transition is good.
I guess I also have this now from rpi-imager telemetry:
https://rpi-imager-stats.raspberrypi.com/