guysoft/FullPageOS #304: Shutdown on boot

SloCompTech, 2019-11-15:
#### What were you doing?
Installed fresh FullPageOS version 0.10.0. Raspberry (3B+) booted first time. When booted it automatically shutted down. Now it doesn't boot again.

#### What did you expect to happen?
Boot normally and open web page.

#### What happened instead?
Raspberry Pi shuts down.

#### Was there an error message displayed? What did it say?

Just showing Started ...
Then just starts showing Stopped ...

#### Version of FullPageOS?
0.10.0

guysoft, 2019-11-28:
Can you try replashing? I am not sure if the image was flashed correctly to start with.

SloCompTech, 2019-12-05:
I will try again, and report.

SloCompTech, 2019-12-11:
Tried again:
- Downloaded version 0.10.0 again
- Written image with balena etcher to SD card
- booted up Raspberry Pi 2B+
- it booted up, but just after booting up, it shuted down automatically
[image]

guysoft, 2019-12-12:
That screenshot is unhelpful, it just shows a showdown process.
There is something triggering a shutdown - if you just flash plain Raspebian does this happen too?

SloCompTech, 2019-12-14:
No if I flash normal raspebian it works

guysoft, 2019-12-15:
Ok, can you provide a fuller boot log in any way? I really can't tell what is going on.
I am just not sure how to debug this? Perhaps a video of the whole boot sequence?

SloCompTech, 2019-12-15:
Hi,
sorry it is model 2B, here is `/var/log/syslog` https://pastebin.com/0ig0tc5q

guysoft, 2019-12-17:
Might be this error:
```
Dec 11 18:50:36 fullpageos lightdm[448]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
```

What is in ``/etc/lightdm/lightdm.conf``?

Also, did you try 0.11.0 RC1? might be solved there. it looks like lightdm is having issues.

SloCompTech, 2019-12-21:
Hi,
just tried 0.11.0, no luck, still no audio output.

guysoft, 2019-12-22:
@SloCompTech You didn't mention anything about audio output until now, in fact, this makes no sense because you said in the issue that the gui shuts down right away - have you solved it? If not, how are you testing the audio?

SloCompTech, 2019-12-22:
Ouch, to much stuff in my head. I meant video output after boot. RPi boots up then there is no video signal on HDMI (monitor does not detect signal).

guysoft, 2019-12-22:
@SloCompTech but you posted the screen here: https://github.com/guysoft/FullPageOS/issues/304#issuecomment-564683131

Did you switch hardware?  Is the system working with other monitors?

SloCompTech, 2019-12-22:
I will try that.

SloCompTech, 2020-03-12:
Now working with RPi2B with version 0.11.0 build 26/9/2019.

closed 2020-03-12 by SloCompTech

guysoft, 2020-03-13:
@SloCompTech Did you do anything to fix it? Or just reflashed?

SloCompTech, 2020-03-13:
Just flashed version 0.11.0, nothing modified.