guysoft/FullPageOS #305: FullPageOS 0.11.0 RC1 status

guysoft, 2019-12-02:
FullPageOS 0.11.0 first release candidate.

Features:
* Chromium Version 74 (Official Build)
* Debian Buster
* Support for RaspberryPi 4 B

Please test and report if there are any problems

## Download
http://unofficialpi.org/Distros/FullPageOS/nightly/2019-06-17_2019-04-08-fullpageos-stretch-lite-0.10.0.zip
md5: [``0ae6d01799639fcedcdacb97b05284b7``](http://unofficialpi.org/Distros/FullPageOS/nightly/2019-06-17_2019-04-08-fullpageos-stretch-lite-0.10.0.zip.md5)

## no-acceleration variant

http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2019-12-01_2019-09-26-fullpageos-no-acceleration-buster-lite-0.11.0.zip
md5: [``1e82711d8f5d85998ca03c85f263396d``](http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2019-12-01_2019-09-26-fullpageos-no-acceleration-buster-lite-0.11.0.zip.md5)

guysoft, 2019-12-09:
URL was fixed, I guess no one has tested it since I got no replies since. I am just gonna test it and release.

xbeaudouin, 2019-12-31:
Works for me :D (on RPI 4).
Didn't try yet on RPI3, have issue with official rpi display...

guysoft, 2019-12-31:
Released
https://github.com/guysoft/FullPageOS/releases/tag/0.11.0

closed 2019-12-31 by guysoft