guysoft/FullPageOS #306: Can't hide mouse coursor

Szwajowy, 2019-12-06:
#### What were you doing?
I followed this steps:

> edit lightdm.conf found in /etc/lightdm/lightdm.conf and look for
> 
> #xserver-command=X
> change that to
> 
> xserver-command=X -nocursor
> reboot and cursor should not appear

#### What did you expect to happen?
cursor should not appear

#### What happened instead?
cursor appeard

#### Was there an error message displayed? What did it say?
no errors

#### Version of FullPageOS?

2019-09-26-fullpageos-no-acceleration-buster-lite-0.11.0

guysoft, 2019-12-11:
Hey,
I am not sure how to fix that on FullPageOS side, seems like a lightdm bug, you shoud file it at https://launchpad.net/lightdm

Szwajowy, 2019-12-15:
I found solution !
You need to edit this file:
/usr/share/lightdm/lightdm.conf.d/*.conf
instead of one mentioned in wiki.

guysoft, 2019-12-15:
If there is a way to make it hide when its not moving I'd add that to the distro

Szwajowy, 2019-12-15:
I'm not asking you to change it on dostęp, but to change it on the Wiki section.

guysoft, 2019-12-16:
@Szwajowy Oh, that was donated, you can add it there yourself, its a wiki :)

If you are having issues you can tell me exactly what to change and I can edit it.

Szwajowy, 2019-12-16:
Ohh, okay. 
Thanks for help :)

closed 2019-12-16 by Szwajowy