guysoft/FullPageOS #307: Chrome Extension: "Virtual Keyboard" working in Kiosk-Mode with normal RasbianStretch

strunx, 2019-12-16:
the Chrome Extension "Virtual Keyboard"

https://chrome.google.com/webstore/detail/virtual-keyboard/pflmllfnnabikmfkkaddkoolinlfninn?hl=en

is working fine with normal Rasbpian Stretch and Chromium 72 autostarted in Kiosk-Mode

but not with that fullpageOS-version:

2019-06-17_2019-04-08-fullpageos-no-acceleration-stretch-lite-0.10.0

There must be something different in system-configurations, but i dont find it.

some ideas?

guysoft, 2019-12-17:
You might want to try the release candidate that uses also Chromium 72.
https://github.com/guysoft/FullPageOS/issues/305

asdf1nit, 2020-02-10:
I use this same extension and have no issue once it's configured.

Linkek, 2023-10-07:
I created a project to trigger the matchbox-keyboard by a custom Chrome extension and a Node server. 

https://github.com/Linkek/keyboardTriggerServer

guysoft, 2023-10-22:
@Linkek Thats really cool! Do you want to ship this somehow on FullPageOS Or have it documented in the FAQ?

Linkek, 2023-10-24:
@guysoft Sure! We could do that. Might need to test it out some more.

csabal, 2024-02-11:
> I created a project to trigger the matchbox-keyboard by a custom Chrome extension and a Node server.
> 
> https://github.com/Linkek/keyboardTriggerServer

Thank you very much! I was able to install and configure it to my need as a museum!!!!! So happy.

guysoft, 2024-02-18:
@csabal Neet, send a photo if you can :)
I love to see where FullPageOS ends up.