guysoft/FullPageOS #308: Dashboard issues

Notyouagain, 2019-12-18:
#### What were you doing?

Trying to install the nightly build of 0.11.0 from 12-1-19 to build a fullpagedashboard with 4 different site tabs.

#### What did you expect to happen?
I expected my tabs to open the pages

#### What happened instead?
Most of the web sites I want to visit fail to load, I believe this is related to iFrames of the Dashboard.  The Default Welcome tab and configuration page suggest there is an experimental version which does not use iFrames exists.  I sent a join request and have not heard back.  I am hoping someone can provide a download link.

#### Was there an error message displayed? What did it say?
No

#### Version of FullPageOS?
0.11.0 from 12-1-19

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2019-12-19:
Can you provide the URLs? Are they using https? Or http?

Notyouagain, 2019-12-19:
https://www.andersoncolumbia.com
https://www.psp.fmcsa.dot.gov
https://clearinghouse.fmcsa.dot.gov
https://hosdvr.trimblefsm.com/efleetsuite/login.aspx

I believe it is worth mentioning the forth link works in a dashboard tab, all the others will not.

guysoft, 2019-12-19:
BTW the first one does work.
The second gives this:
```
Refused to display 'https://clearinghouse.fmcsa.dot.gov/' in a frame because an ancestor violates the following Content Security Policy directive: "frame-ancestors 'self'".
```
The way I extracted this message is adding ``--remote-debugging-port=9222`` to ``/home/pi/scripts/start_chromium_browser`` to the first chromium command there. Then using ssh port fowarding I could see the chromium console using ``ssh fullpageos.local -L 9222:localhost:9222``.

It means that the website has a special http header which blocks it form being displayed in other iFrames.

This extension solves the https issue the issue:
https://chrome.google.com/webstore/detail/ignore-x-frame-headers/gleekbfjekiniecknbkamfmkohkpodhe
However, you can't enable it for kiosk mode because the button is missing.
If I install it and refresh FullPageDashboard seems to go crazy.

Another solution attempt I tried was to add to your chromium start command `` --disable-web-security --disable-site-isolation-trials``
However, that makes me get 
```
Refused to display 'https://clearinghouse.fmcsa.dot.gov/' in a frame because an ancestor violates the following Content Security Policy directive: "frame-ancestors 'self'".
```