guysoft/FullPageOS #311: Chromium stoped autostarting

Szwajowy, 2020-01-12:
#### What were you doing?

I installed a apache server,
cleaned var/www/html
and added my website.
Also changed the link in fullpageos.txt to point to http:localhost

#### What did you expect to happen?

I expected it to boot my site at start in chromium.

#### What happened instead?

It did work on first try, then I updated my site, and it broke. Now after booting there is only a black screen, but apache still works because I can see it on other devices in local network.

#### Was there an error message displayed? What did it say?

There was not.

#### Version of FullPageOS?

0.11.0

guysoft, 2020-01-13:
It needs to be ``http://localhost``, you missed the double-slahes.
Please provide your full ``fullpageos.txt``

guysoft, 2020-01-14:
Closing if no reply

gbrakcounty, 2021-06-01:
I am having the same issue but my /boot/fullpageos.txt has a valid URL.  I can run bash ~/scripts/start_chromium_browser which works fine and loads the desired page.  I checked to see if chromium is running at startup and it is not.  I just have a black screen.

closed 2021-06-01 by guysoft