guysoft/FullPageOS #313: Camera permission setting

jyy1082, 2020-01-21:
I tried to run Webrtc page on FullPageOS with a raspberry pi 4, but can not get camera access permission. Is there someone know how should we set the camera and microphone permission for FullPageOS? Thank you!

dan-developer, 2020-06-09:
Up

guysoft, 2020-06-09:
Using rpi-config?
How is the website accessing the camera?
Please be more specific - how is this different than using raspbian?

cweiske, 2020-07-22:
`start_chromium_browser` runs chromium with `--kiosk` which probably disables the dialog.

guysoft, 2020-07-22:
[According to this](https://stackoverflow.com/questions/17069979/how-can-i-make-chrome-allow-access-to-a-webcam-over-http-not-https) you can override it with: 
```
mkdir -p /etc/opt/chrome/policies/managed
touch /etc/opt/chrome/policies/managed/test_policy.json
```

In ``test_policy.json``:
```
{
  "VideoCaptureAllowedUrls": ["http://www.example.com/", "http://[*.]example.edu/"]
}
```