guysoft/FullPageOS #314: Pass Arguments to Chromium upon startup

tmello01, 2020-01-25:
Is there a way to pass arguments to the chromium application before it starts up? For example, I need the browser to be run in kiosk mode, with disabled pinch-to-zoom, and disabled over-scroll history change. These can all be disabled by passing various arguments to the chromium-browser when it starts up, is there a way I can configure these parameters?

guysoft, 2020-01-26:
Three of those settings are the default set in the distro 
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/start_chromium_browser#L11
You can edit the script to add more or remove settings. 
When you say "configure them as parameters?" Where would you actually configure them?

stappersg, 2020-10-20:
@tmello01 see #316 and please consider closing this issue.

guysoft, 2020-10-28:
Closing due to no reply.
Since an answer was given on Jan

closed 2020-10-28 by guysoft