guysoft/FullPageOS #316: Fullpageos Proxy settings

lukeyboytas1, 2020-02-09:
HI - Any help on how to add proxy server settings?

guysoft, 2020-02-16:
If you press ctrl+t on a keyboard/vnc you should get a nornal chromium window where you can access the browser settings and set up a proxy

lukeyboytas1, 2020-02-16:
That is an awesome start.... Now I get this ( see pic)
[image]

guysoft, 2020-02-16:
Oh, seems like when you paste that message in to google you get a commandline argument to do this ([source](https://www.internalpointers.com/post/run-chrome-chromium-proxy-settings-linux)).
So it seems you will need to add this argument in the file [``/home/pi/scripts/start_chromium_browser``](https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/start_chromium_browser)
The command arguments are set is this file and you can change them, including adding that proxy server flag.

stappersg, 2020-05-29:
@lukeyboytas1 how about any feedback on the additional advice from @guysoft ?

guysoft, 2020-06-01:
Ill close this issue if there is no reply

lukeyboytas1, 2020-06-01:
Sorry guys - this is over my head - I dont understand adding arguments?

guysoft, 2020-06-02:
@lukeyboytas1 There is a file with a line that writes the command used to start chromium. That command has list of arguments. [Here is an article on this in the chromium project](https://www.chromium.org/developers/how-tos/run-chromium-with-flags).
If you edit that file you can change that line to have different settings. As explained in the chromium tab you took a photo on.
 You need to have basic understanding how to use a raspberrypi to edit a file there. Once you do it should use the proxy from that line that you edit.
The path of the file in the filesystem is ``/home/pi/scripts/start_chromium_browser`` as I provided above.

stappersg, 2020-07-26:
> Oh, seems like when you paste that message in to google you get a commandline argument to do this ([source](https://www.internalpointers.com/post/run-chrome-chromium-proxy-settings-linux)).
> So it [...]

Yes, adding ` --proxy-server="protocol://hostname:port" ` to the many options on `chromium` in `/home/pi/scripts/start_chromium_browser` works for me.

stappersg, 2020-08-24:
@lukeyboytas1 thanks for raising the question.  I'm glad found this issue.  **and** the advice from guysoft.

For is it fine that this issue gets closed.  (@lukeyboytas1 you have to power to do so)

60999, 2021-02-20:
--proxy-server="socks5://127.0.0.1:1080" 
 --proxy-server="http://127.0.0.1:1081"

60999, 2021-02-20:
in
`--proxy-server="protocol://hostname:port"`
<protocol>could be one of them：

"http", "socks", "socks4", "socks5".

stritti, 2023-09-21:
Proxy could also be defined global in Rasbian, which also helps to run apt updates:

```sh
cd /etc/apt/apt.conf.d
sudo touch 10proxy
sudo nano 10proxy
```

```sh
//file=/etc/apt/apt.conf.d/10proxy
Acquire::http::Proxy "http://127.0.0.1:8080";
Acquire::https::Proxy "http://127.0.0.1:8080";
```