guysoft/FullPageOS #318: 2020-02-09_2020 nightly - cannot connect to network via WiFi

bshepps1985, 2020-02-17:
Wanted to run the latest nightly build: 2020-02-09_2020-02-05-fullpageos-buster-lite-0.12.0

Used balenaEtcher to write image to SD card. Edited the fullpageos and fullpageos-wpa-supplicant files accordingly.

The WiFi connectivity does not work.

I am 100% certain the settings are correct.

The 2020-02-05_2020 nightly works without issues with the same settings. 

I am running a Raspberry Pi 4. 

Just an FYI, don't know where else to post this.

guysoft, 2020-02-17:
Hey, you are quite right.
Because rpi foundation changed something without stating in the release notes. I opened an issue here:
https://github.com/RPi-Distro/pi-gen/issues/385

And added a workaround: 
https://github.com/guysoft/CustomPiOS/commit/ed9e7375065c3866564ca87aabb18cbb377c0516

You can fix your current installation by running ``rfkill unblock wifi``. Should be fixed next nightly.

Also please say you are having this too in the issue at: https://github.com/RPi-Distro/pi-gen/issues/385

guysoft, 2020-02-17:
Issue has been solved, closing.

closed 2020-02-17 by guysoft