guysoft/FullPageOS #319: Chromium Update notification

DL6JJ, 2020-02-17:
#### Chromium Update notification
Chromium shows an unexpected notification. It persists even, when it was updated manualy. According to the chromium docs, this behavior should not occur under Linux.

Some ideas?

It is really frustrating, because it occurs seemingly random across devices.

### Looks like this: 
[image]

(The text says: Chromium couldn't update to the latest version, so you're missing out on new features and security fixes.)

guysoft, 2020-02-17:
Has been fixed in #317 pull request.
You can fix the workaround by adding ``--disable-component-update`` in the startup command. There is a new nightly coming out with this fix.

mschmidt291, 2020-02-17:
Actually strange. disable-component-update is not always disabling this message - don't know why.
`--simulate-outdated-no-au='Tue, 31 Dec 2099 23:59:59 GMT'` does the trick and fixed it for us! :)
@DL6JJ

Dragon0501, 2020-02-25:
Where does this lines have to be added?

DL6JJ, 2020-02-25:
`/home/pi/scripts/start_chromium_browser` in line 6 or 11. Depends which case you use.

You can find the original file [here](https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/start_chromium_browser)

@Dragon0501

Dragon0501, 2020-02-25:
Thank you very much

guysoft, 2020-03-01:
I am closing this since it has been fixed and should be available in the current nightly.

closed 2020-03-15 by guysoft