guysoft/FullPageOS #320: Add tor_ssh.sh?

NickBusey, 2020-02-22:
I created a small script a while back that sets up ssh via TOR hidden service when it's run on a machine.

I recently set up FullPageOS and ran the script on it and now I have super easy remote access from anywhere, no messing around with IPs/firewalls/etc.

I figured this might be a useful thing to include. I'm not sure how best to include it in the package, maybe as just an optional setup step in the documentation.

The script is [tor_ssh.sh](https://gitlab.com/grownetics/devops/blob/master/tor_ssh.sh) here's a [short blog post explaining it](https://nickbusey.com/code/2019-03-01-tor-ssh/).

guysoft, 2020-02-26:
Thats pretty neat. I could add it as a module to CustomPiOS. Or just ship the script under "/home/pi/scripts".

You need tor active on your machine in order to reach that.address,  right?

guysoft, 2020-04-19:
Been playing with it, its neat, but beyond providing an install script and shipping it in ``/home/pi/scripts`` I am not sure how to accommodate it.

NickBusey, 2020-04-19:
That sounds like a decent option. Or maybe just a note in the docs about how to run the script and why you might want to do so.