guysoft/FullPageOS #321: Chromium load with changed resolution on rotated display

derSebastian, 2020-03-01:
#### What were you doing?
Rotated display but and resolution set to 1920x1080. Modified config.txt to various modes aas well as via sudo raspi-config.

Tried:
hdmi_group=1
hdmi_mode=16 (and 34)

also
hdmi_group=2
hdmi_mode=82

rotated the display via
display_rotate=3
and also tried via:

display_hdmi_rotate=3
or
lcd_rotate=3

I have these settings on a RP2 with those settings and that worked.
I have also tried both the legacy drivers and the Fake KMS drivers via raspi-config

When login in via VNC I can see the horizontal dimply fully. This should be vertical now.

#### What did you expect to happen?
That Chromium loaded and would now have a full screen vertical appearance. This is something I have running on an older RP2

#### What happened instead?
The boot up sequence has the correct resolution and orientation, but when booted and loading chromium / kiosk mode (the web page via FullPageOS.txt) the resolution switches to 1360x768 and the display stays rotated but the browser is not. See image

#### Was there an error message displayed? What did it say?
Nope

#### Version of FullPageOS?
0.11.0

#### Screenshot(s) showing the problem:
[image]

guysoft, 2020-03-01:
I don't really have experience with this and its more of a raspberrypi/xorg thing.
But you might take a look at this:
https://www.raspberrypi.org/forums/viewtopic.php?t=248498

And might be related to this bug:
https://github.com/guysoft/FullPageOS/issues/137

Try switching off the acceleration or using the non-acceleration variant.

asdf1nit, 2020-03-24:
@derSebastian I've tried to do the same and have come up with a hack to the solution but I'm not really satisfied. 

you can change the display to rotate the way you want by logging in as pi via ssh and using the following:
`
export DISPLAY=:0
xrandr -o left
`
This rotates the display correctly but I'm looking for a way to make this change permanent without having to run the command every time. For now I have edited the fullscreen script in the pi/scripts directory to run `xrandr -o left` right after the `export=:0` that's already there and it works but it seems like this should live in the xorg.conf file from what I've read.

I have yet to find exactly where in the configuration this file is, or how I would create the correct file in `/usr/share/X11/xorg.conf.d/` directory. If anyone knows please advise and I'll create a pull for adding this to the project.

asdf1nit, 2020-04-19:
After some playing around and getting this on a test system I found that while my method worked it did not rotate the mouse/touch screen orientation with it. I finally got some time to get back to this to find or create a solution for this and I came across a gist that was great. 

https://gist.github.com/mildmojo/48e9025070a2ba40795c#gistcomment-2694429

Dropping this script on my test system worked great so if you have the same issues give it a try. You will have to install xinput on the system and with the pi user run `export DISPLAY=:0` before you run the script. I have not tested this on a pi3 yet but I see no issue there. While this still doesn't solve screen rotation/orientation being set in the boot config it does work quite well and I'm satisfied with the results.

@guysoft I'm thinking this would be nice to have in the system as a script that a user could run but am wondering what the best way this should be included. Before I submit a pull I wanted to get your feedback as I see a few possible ways it could be added.

1. Add the script to FullPageOS. While just being another script in the script folder would be nice it does depend on the xinput package, which isn't on the system by default, and it would still be up to a user to configure how they run the command. Adding the script and xinput to the build would be the easy way to just let users configure the system how they want at a later date.

2. Add the script to CustomPiOS. If there's overlap and users there would benefit as well it could go on the base there instead of in FullPageOS

3. Add it as a Module. Standalone module or include it with the Admin-Toolkit, either way it could give options for setting things up at build time and not having to modify each system separately.

guysoft, 2020-04-21:
@asdf1nit I would vote for either 1 or 2. I think 2 is s good idea, you could add it to the ``gui`` module. The path is here: https://github.com/guysoft/CustomPiOS/tree/devel/src/modules/gui/filesystem/home/pi/scripts

closed 2020-05-06 by guysoft