guysoft/FullPageOS #322: Add documentation to README for changing Splash Screen

tmello01, 2020-03-12:
In the README for this project, it is stated as a feature that you are able to **"Specify a custom Splashscreen that gets displayed on booting process instead of Kernel messages/text"**, but there is no documentation on how to do this in the README or the Wiki for this project. Can we add these steps somwhere?

guysoft, 2020-03-12:
You are quite right. That was part of PR #317 done by @mschmidt291 

It can be changed by editing the file ``/boot/splash.png``.

@mschmidt291 are there any constraints on the file?
It might be worth adding a note in the README and redirecting people [to the wiki with a guide](https://github.com/guysoft/FullPageOS/wiki)

mschmidt291, 2020-03-12:
Yes, just replace /boot/splash.png. But for this to work, you have to build fullpageOS yourself, since the releases in this repo don't have my commit included yet. 
@guysoft any plans on releasing a new version with included Splashscreen? I then can add documentation to the README, but I think the majority of people is using your pre-built images, instead of compiling it.

guysoft, 2020-05-30:
@mschmidt291 Hey, I do have plans. The RaspberryPi Foundation just released a new Pi and renamed Raspbian to Raspberry Pi OS, and changed all the link names. I want to get that working and tested, then we can release.

nabheet, 2024-12-01:
Splash screen is not showing up on my RPi Zero W. 

I added my image to both `/boot/splash.png` and `/home/pi/media/splash.png` from https://github.com/guysoft/FullPageOS/pull/317/files#diff-651e9d987beed945f3e80b5f4d3462a949e3074bf6b70383c6a19e21b9696d7cR31.

not sure what else to try.

guysoft, 2024-12-01:
@nabheet RPi Zero W is not supported.
https://github.com/guysoft/FullPageOS?tab=readme-ov-file#requirements

RomanSmolka, 2025-02-28:
The following is works:

- Put the image into `/boot/firmware/splash.png` – this is for the bootloader
- Change line 5 in `~/scripts/run_onepageos`
from `feh --bg-center /opt/custompios/background.png`
to `feh --bg-center /boot/firmware/splash.png`
– this is for the startup script

peteruithoven, 2026-04-25:
I've added a wiki page:
https://github.com/guysoft/FullPageOS/wiki/Change-Splash-Screen

Replacing both images seemed simpler and easier to script that editing `~/scripts/run_onepageos`.

closed 2026-05-17 by guysoft