guysoft/FullPageOS #324: Page not displaying correct resolution

shock5006, 2020-05-12:
#### What were you doing?

FullPageOS keeps displaying in 1360x768

#### What did you expect to happen?

1920x1080

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

0.11.0

Changes to config.txt seem to only effect the terminal screen. Once Chromium loads, it always goes into 1360x768. There are no resolution options in raspi-config, it just says 'default monitor preferred resolution'

I can confirm that the screen is definitely capable of 1080p

guysoft, 2020-05-13:
What are you setting in ``config.txt``? Please provide the file. As far as I can tell its expected behaviour, if its not its a Raspbian issue. This is the best I can guess since you did not provide any logs.

shock5006, 2020-05-13:
I'm using the default `config.txt` that comes with FullPageOS
I've tried setting every option I can find that sets the resolution and none of them work.
I could post the `config.txt` if you want, but it's just the default one from the `.img`

guysoft, 2020-05-13:
Ok, does this happen in raspbian (desktop version)?

shock5006, 2020-05-13:
Yes it defaults to the lower resolution, but I am able to change it in
display settings.

As I can't access the display settings when FullPageOS is running, it's
stuck on the lower resolution.

guysoft, 2020-05-13:
Ok, sounds like its a raspbian issue. Regarding the defaulting you can an issue there.
You can access the display settings of FullPageOS using either a keyboard and entering a virtual terminal. Or using SSH and providing the environment variable ``DISPLAY=:0`` and running [xrandr](https://wiki.ubuntu.com/X/Config/Resolution). If its not installed you can ``apt-get install`` it.

shock5006, 2020-05-13:
Ok great. I've left work for the day so I'll give it a go tomorrow and let
you know how I go. Thanks.

guysoft, 2020-05-13:
Ok, also friendly reminder that this is not my work, this is an open source project I am 100% doing in my spare time. So if this benefits your work and give you money, please consider making a donation, there is a link in the README main page on the github. Hope we figure this out.

shock5006, 2020-05-14:
Hi again. I tried your suggestion. xrandr seemed unable to detect the HDMI output, so I couldn't apply a custom resolution to it.
So I gave up and swapped it with another panel, and it works perfectly.
I'm pretty sure the TV just had a faulty EDID or something.
Thanks for all your help. I'll see about making a donation asap.

guysoft, 2020-05-14:
Which panel BTW? might help future people having this issue.

Murugesh-Hobbyist, 2024-06-12:
Hi @guysoft, firstly appreciating your work.
I too facing the screen resolution problem. my screen looks like zoomed and cropped.
If I burn fullpageos on pi-3, it works great, also no bootup texts.
but the pi-4 image is causing this crop issue and also shows bootup texts.

`
display@twinbot:~ $ sudo apt-get install xrandr
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package xrandr is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  x11-xserver-utils
E: Package 'xrandr' has no installation candidate

display@twinbot:~ $ xrandr
Can't open display
display@twinbot:~ $ xrandr -s 1024x768
Can't open display
`

guysoft, 2024-06-13:
@Murugesh-Hobbyist Sounds like a specific monitor/device paring issue - what happens if you try a different monitor?

Murugesh-Hobbyist, 2024-06-13:
Ohh ok. Will try it out and update here @guysoft

Murugesh-Hobbyist, 2024-06-27:
Hi @guysoft Tried different monitor works great. thanks!

Murugesh-Hobbyist, 2024-08-18:
Hi @guysoft I have recently upgraded using below comand.
sudo apt update
sudo apt upgrade

Now the display is autochanged to 4:3 ratio.
In my monitor I have option to switch to (AUTO , 4:3 , WIDE)
1. If I use AUTO/4:3, Diplay is perfectly fits as 4:3 of monitor without stretching and shows left and right long vertical black borders.
2. If I use WIDE, Display stretches landscapely with improper ratios.

I couldn't follow these options as not exists:
[](https://pimylifeup.com/raspberry-pi-screen-resolution/
)
I request you to suggest a way to set the display resolution to 16:9 instead of 4:3

guysoft, 2024-08-18:
@Murugesh-Hobbyist sometimes certain monitors don't play well with a rpi on a newer version. I can't support upgrades, I have no infrastruactire and time to do that. I can try help you find what is going on - what happens when you plug a different monitor?

Murugesh-Hobbyist, 2024-08-18:
ohh okay. I tried to plug other monitor, it also sticks with 4:3 ratio.

guysoft, 2024-08-18:
I have no idea why it would do that. Perhaps someone here might.

frutabruta, 2024-08-18:
I had a similar problem with a different board (BTT CB1). The display has to communicate with the board by I2C lines (included in the HDMI cable) to check the resolutions supported by the display. If the I2C lines are broken somewhere along the way (bad soldering on the Raspberry connector, cable, microHDMI adapter), the Raspberry might default to 1024 × 768, which is a 4:3 ratio.