guysoft/FullPageOS #325: Error in line 22 sed command in start_chroot_script

tjuuljensen, 2020-05-13:
Last week, I installed the latest nightly build and I really liked the new custom logo option.
There is a problem though in FullPageOS/src/modules/fullpageos/start_chroot_script
(guess it's a typo?)

Line says:
sed -i 's/n/ logo.nologo consoleblank=0 loglevel=0 quiet/' /boot/cmdline.txt
This is a wrong use of sed and breaks the command "console=" in two.

Instead it should say:
sed -i 's/$/ logo.nologo consoleblank=0 loglevel=0 quiet/' /boot/cmdline.txt

/tjuuljensen

guysoft, 2020-05-13:
Seems, so git-blame shows its from commit https://github.com/guysoft/FullPageOS/commit/655d00562a9b7af5e61956f5b0e734195b4ccdf0 done by @mschmidt291 on Feb.
Fixing it unless you want to send a PR and have the next git-blame :)

guysoft, 2020-05-13:
Building another nightly

closed 2020-05-13 by guysoft

tjuuljensen, 2020-05-14:
Cool.  Thanks 👍🏻

commit 8e33f06d (2020-05-13): Fixes #325 introduced at #317