guysoft/FullPageOS #326: Debug FPOS not getting past splash screen

tmello01, 2020-05-26:
For some reason, FPOS now stops at the splash screen. There is no chrome log or any other systemctl unit failure from what I can see, it just does not continue. 

Manually stopping the splashscreen service opens the boot log, but there seems to be no failure that I can see, rebuilding and reinstalling does not help.

guysoft, 2020-05-26:
1. What version of FullPageOS?
2. Have you chaged anything?
3. Has this been running for a while or is it a freshly flashed image?
4. Also what do you mean by "rebuilding"?

You ignored the template, which means I have no information on what you are doing.

tmello01, 2020-05-26:
1. Using the latest on the repository
2. Edited the start_chroot_script to install a package using apt.
3. It first happened on an image running for a while, but I tried reinstalling with the same ISO, chrome still would not start. 
4. I have a Ubuntu distro running a build on the OS following the instructions in the README. It was working previously but just stopped for some reason. 

Sorry about the template thing, I was more looking for logs or other debugging info but I'll take any help I can get!

guysoft, 2020-05-26:
1. What version, there are nightly builds and stable builds, both are latest. Please be specific. Explain exactly what you are doing.
2. Are you using the build script to build an image or are you using the official builds? If you are building please provide the ``build.log`` file of your build "What were you doing?" . I can't guess what you mean by "it does not work". Tell me all the steps you are doing so I can begin to understand the context. If you are editing ``start_chroot_script``, then please post what changes you made.
4. Which instructions in the readme? are you using the docker build method or any other method? The Docker build method is the recommended way to build in order to avoid environment issues.

tmello01, 2020-05-29:
I figured out the issue. I added some lines to the start_chroot_script to disable the startup boot text, removing that fixed the issue. Is there a way you suggest to hide the text that appears before the splash screen?

guysoft, 2020-05-29:
Can you please share your code so other people might benefit from your attempts?

tmello01, 2020-06-01:
No problem. 

Following online guides (Found [here](https://retropie.org.uk/forum/topic/14299/tutorial-remove-boot-text-on-the-raspberry-pi-for-noobs/3)) I added the following lines to the start_chroot_script file:

`sed -i 's/console=tty1/console=tty3/g' /boot/cmdline.txt`
`sed -i 's/loglevel=0/loglevel=3/g' /boot/cmdline.txt`

While the sed commands are successful and the log messages hidden, doing this causes the desktop never to show, instead to be stuck on the Splash Screen forever. 

My full boot/cmdline.txt file looks like this after the sed commands:
`console=serial0,115200 console=tty3 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait logo.nologo consoleblank=0 loglevel=3`

guysoft, 2020-06-01:
It might be that somehow its clashing with the recently added splash screen, thats a new addition to the nightly builds. https://github.com/guysoft/FullPageOS/pull/317

You can try disabling the splash screen, or use the stable version of FullPageOS and see if that also solved the issue. If it does then its important for me to know two because I am planning to release soon and this is a new feature added. 
For example there is a talk about release go-no-go here: 
https://github.com/guysoft/FullPageOS/issues/322#issuecomment-598233749

I guess sharing does not only help user, it also helps me as a maintainer understand better what people are doing.

tmello01, 2020-06-01:
Thanks for the help. You were right it is an issue with the splash screen setup the way it is now. I figured out a way to do it that now hides all of the boot messages but still displays the splash screen as expected. Would you rather me post a detailed guide or create a PR with the changes so it can make it into the next build?

guysoft, 2020-06-02:
In a way I want both.
What exactly can you add in the PR that would solve the issue? It sounds like two settings that clash with no workaround?
If you can solve it without making them clash then please do PR :)

tmello01, 2020-06-02:
So the way I solved it was to utilize plymouth instead of FBI like FPOS currently is. Plymouth is able to display the splash screen without affecting the boot messaging settings. I suspect this is because FBI requires STDOUT to display it's image which, when changed to hide boot messages, causes some weirdness.

So I guess my solution is less solving the clashing issues and more trying to remove one of the things causing the clash. Does this work for you?

guysoft, 2020-06-02:
Thats a really good PR. If you can make it I will merge it.
The code for that is here:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/start_chroot_script#L18

Also, if its all good you are also welcome to add this as a feature to the gui module for CustomPiOS (that might take more ping pong between us because it works on several distros).
That way any distro that uses the GUI will have a splash screen.

The gui module is here: 
https://github.com/guysoft/CustomPiOS/tree/devel/src/modules/gui