guysoft/FullPageOS #327: "Can't update Chromium" message inside browser

papierkorb27, 2020-06-03:
Running the kiosk-pi with FullPageOS I got the message "Can't update Chromium" inside Chromium some hours after booting. That seems to be an annoying Chromium bug. Maybe its interesting for FullPageOS users to know this workaround until the bug will be fixed: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=264399&start=50

`sudo touch /etc/chromium-browser/customizations/01-disable-update-check;echo CHROMIUM_FLAGS=\"\$\{CHROMIUM_FLAGS\} --check-for-update-interval=31536000\" | sudo tee /etc/chromium-browser/customizations/01-disable-update-check`

guysoft, 2020-06-09:
You are welcome to make a pull request

markgaze, 2020-06-15:
This was fixed in #317 ?

guysoft, 2020-06-15:
@markgaze That issue is unrelated, its about the splash screen before chromium starts.

markgaze, 2020-06-15:
@guysoft whilst that is true, it also fixed #319 which I think this is a duplicate of? Apologies if this is a slightly different issue!

guysoft, 2020-06-15:
@markgaze Indeed it did, I didn't read though all the issue.
Ok, cool, thanks for helping me tidy up!

closed 2020-06-15 by guysoft

markgaze, 2020-06-15:
> @markgaze Indeed it did, I didn't read though all the issue.
> Ok, cool, thanks for helping me tidy up!

Any time! Thanks for maintaining this project which makes my life much easier! 😁