guysoft/FullPageOS #330: FullPageOS Build Failing due to Package Get Issues

tmello01, 2020-06-08:
#### What were you doing?

Building FPOS via command line

#### What did you expect to happen?

Build would pass and produce an ISO. 

#### What happened instead?
Build is failing on installation of "matchbox-window-manager xorg lightdm" due to checksum issues.

#### Was there an error message displayed? What did it say?
```
Err:26 http://raspbian-us.ngc292.space/raspbian buster/main armhf fonts-dejavu-core all 2.37-1
  Hash Sum mismatch
  Hashes of expected file:
   - SHA256:58d21a255606191e6512cca51f32c4480e7a798945cc980623377696acfa3cfc
   - SHA1:0f2c53b4c24fb2bd236fee3ce38266bb703dea82 [weak]
   - MD5Sum:5f413fbf5c17b6417affa23aafe587ee [weak]
   - Filesize:1068028 [weak]
  Hashes of received file:
   - SHA256:0a5af7daf0dd4bf745777795ccfb1f07d4ebb6d7b69a9493a5ef3170d86ddb8c
   - SHA1:146aab584feac037bc07a631b3326281827db629 [weak]
   - MD5Sum:5ed81529ae8d69ac7e866412e655321d [weak]
   - Filesize:185367 [weak]
  Last modification reported: Mon, 08 Jun 2020 10:44:11 +0000
```
#### Version of FullPageOS?
Latest downloaded from github repo. 
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build
[build.log](https://github.com/guysoft/FullPageOS/files/4745453/build.log)

tmello01, 2020-06-08:
More information: this seems to be an issue with Raspbian's sources. See this stackexchange question:

https://raspberrypi.stackexchange.com/questions/113178/cant-upgrade-or-install-software-get-a-strange-error-that-says-it-failed-to-fe

and also this reddit post:
https://www.reddit.com/r/pihole/comments/gxfu7/i_get_the_error_failed_to_fetch_https113storecres/

Could this require a change to the build process to update the sources before beginning getting the packages?

tmello01, 2020-06-08:
Fixes tried:

1. Downloading latest Raspbian Lite image for building (using instructions in README)
2. Adding SED command to FullPageOS/src/modules/fullpageos/start_chroot_script to replace mirror with different mirror, did not seem to help, as I think this is happening in the base custompios start_chroot script that I can't quite find yet.

guysoft, 2020-06-08:
Hey, checksum issue means that the build is failing to download the image. This does happen from time to time. It means their mirror, looks like the US one from your log, has an issue. It would fail if you run it on a Pi in the same region. You can try switching mirrors. If you have some big production system and can't afford downtime like this you can also make your own mirror. Also in that case I also recommend making a donation here :)

tmello01, 2020-06-08:
Makes sense, it seems that particular mirror is down for everyone. Is there a way I can edit the image to use the new sources? Or can you point me to the top level start_chroot_script so I can add a "sed" to change the host before we do any "getting"?

tmello01, 2020-06-08:
I fixed this issue with the following command:

`sed -i s/http:\/\/raspbian.raspberrypi.org\/raspbian\//http:mirror.us.leaseweb.net\/raspbian\/raspbian\//g' /etc/apt/sources.list` 

Thanks for the help! :)

closed 2020-06-08 by tmello01

guysoft, 2020-06-08:
Ok, note that the url that is shipped with the raspberry pi os points to a node balancer that routes according to your ip.

So if you are forcing a particular mirror it might be really slow for a user using your distro outside the US. You want the routing to go according to their IP.