guysoft/FullPageOS #331: Webcam support

dan-developer, 2020-06-09:
Is there a way to enable webcam support? For example, to run the page: https://pt.webcamtests.com/

guysoft, 2020-06-09:
Do you want enable the camera in a distro you are building?  Or as a user flashing one of the existing images?

guysoft, 2020-06-11:
Ok, sounds like a chromium issue, you can open an issue on the chromium issue tracker: https://bugs.chromium.org/p/chromium/issues/list

dan-developer, 2020-06-11:
On the computer it works normally. In the raspberry the use of the webcam is not requested, and therefore if it is not requested it is not released.

[image]

guysoft, 2020-06-14:
@dan-developer If you do ``ctrl+T`` and open a window that is not in kiosk mode, and then use that window, does it work then?

guysoft, 2020-07-01:
Any news on the workaround I suggested above?