guysoft/FullPageOS #333: How can remove boots messages?

matiasmenares, 2020-06-14:
How i can remove all boot messages, but keeping splash screen?

y change to console=tty3 but don't open chromium.

guysoft, 2020-06-15:
That is not really the scope of FullPageOS.
You could just put a blank splash screen.
This is a good moment to point out - that if you are removing attribution of the software I am offering 100% for free and open source, that I accept donations. I get no money to develop FullPageOS/CustomPiOS. Its all volunteer work.

matiasmenares, 2020-06-17:
I dont want hide the splash screen with de OS image, i just want remove the boot texts start OS, and of course i will donate the project! 🚀

guysoft, 2020-06-17:
Ok, what if you change the splash image to a black image? would that be enough?
What exactly is the use case that you need it so hidden?

matiasmenares, 2020-06-17:
I need hidden this messages

[https://ibb.co/VwT7QM0](url)

guysoft, 2020-06-18:
I don't have an instance at hand - aren't those hidden when using the splash screen?
Before you say you don't want the splash screen - I am trying to establish if there is a mechanism that hides them in the current image you can use, before looking for more complicated solutions.

stappersg, 2020-10-20:
> I need hidden this messages
> 
> https://ibb.co/VwT7QM0

_That page doesn't exist_

Perfect illustration how much original poster cares about the issue.

And that suits me well, I think  it is plain wrong to hide technical information.

Leaving it up to @guysoft to let this issue rot or to close it.

Indeed, if I had the power to close this issue, I would have already done it.
Main motivator is preventing that _no one cares about issues_ drain human energy.

stappersg, 2020-10-20:
@matiasmenares does #322 honor your wish?

matiasmenares, 2020-10-26:
@stappersg this boot messages https://i.stack.imgur.com/2uAWG.jpg