guysoft/FullPageOS #338: Touch Controls are inverted on x axis

conrad9900, 2020-06-26:
#### What were you doing?

Booting to a HDMI touchscreen

#### What did you expect to happen?

Normal touchscreen actions

#### What happened instead?

up and down touch is ok, left and right is inverted

#### Was there an error message displayed? What did it say?

No error messages

#### Version of FullPageOS?

2019-12-01_2019-09-26-fullpageos-buster-lite-0.11.0
--
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2020-06-26:
What type of touchscreen?

guysoft, 2020-07-01:
Any news on this? Is this the official touchscreen or something else?

conrad9900, 2020-07-01:
> Any news on this? Is this the official touchscreen or something else?

No, it is not official. It is a 15" POS, uses eGalax Driver.
I have tested several settings and i have concluded that the chinese screen just isnt worth the hassle.
This isssue can be closed

closed 2020-07-01 by conrad9900