guysoft/FullPageOS #340: Possible to use Firefox instead of Chrome

giddel, 2020-07-08:
I've websites which cannot be viewed with the delivered Chromium browser due oauth reasons. But a manually installed Firefox ESR  can display this sites properly. Unfortunately  it lacks the kiosk mode in the delivered version 68. But since Firefox v71 there is a kiosk mode available too. 
So my question: would it be possible to deliver FullpageOS with an optional current Firefox browser?

guysoft, 2021-11-01:
Could be done with https://github.com/OctoPrint/CustoPiZer or CustomPiOS, but I won't maintain it. You could write a CustoPiZer version that would run on GitHub actions.

nekromoff, 2024-07-08:
For anyone replacing chromium (crashes often, so I had to use older version chromium-browser_88.0.4324.187-rpt1_armhf.deb anyway, but that sometimes crashed too) with firefox-esr:

1. replace ~/scripts/start_chromium_browser content with `DISPLAY=:0 firefox-esr --kiosk https://example.com`
2. remove execute bit (`chmod -x`) from these files ~/scripts/: refresh, fullscreen

That should be enough. You can have a cronjob checking every minute, if firefox is running like this:
`check_browser.sh`:
```
if pgrep firefox-esr >/dev/null
then
    echo "Process is running."
else
    echo "Process not running, restarting now"
    killall -9 firefox-esr
    rm /home/pi/.mozilla/firefox/USED_PROFILE/.parentlock # remove lock, so we don't get error "Firefox running, not responding."
    logger "firefox-esr: relaunching now"
    DISPLAY=:0 firefox-esr --kiosk https://int.bastadigital.com/overseer/public/NO8wpdIlOURcNewCtFpepyilagFF1BT1
fi
```
change USED_PROFILE to your real firefox profile.