guysoft/FullPageOS #342: Building Image on Ubuntu 20.04 fails

sebaro11, 2020-07-10:
#### What were you doing?

Since the download mirror seems to be down, I tried to build the image on my Ubuntu 20.04 system. But It fails at touching a file.

#### What did you expect to happen?

Building the image.

#### What happened instead?

touching /boot/ssh fails

#### Was there an error message displayed? What did it say?

`+ touch /boot/ssh
/chroot_script: line 63: /usr/bin/touch: cannot execute binary file: Exec format error
`
Using /usr/bin/touch in bash terminal is working all the way.

#### Version of FullPageOS?

0.11.0

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### If you are building FullPageOS - provide a build.log that is created for the build

https://up.ovpn.to/file/ItIc40hlY7mc8yOmcF5OdKsFuj8sNzDwNS8azQyX/build.log

sebaro11, 2020-07-10:
Built it now via ARM Cloud server.

closed 2020-07-10 by sebaro11

guysoft, 2020-07-13:
Want to share how that works?
Also github actions works

Also the mirror is back up.
moreover the images release go though testing.