guysoft/FullPageOS #343: Webpage never fully loads

MODICA-Will, 2020-07-20:
I am trying to load an externally hosted dashboard. It can be viewed publicly at the following URL: https://io.adafruit.com/madebymodica/dashboards/isana-scada?kiosk=true

PROBLEM: With FullPageOS, the background graphics will load, but the actual data graphs/text will not.

What it looks like on FullPageOS:
[image]

What it looks like from other browsers (what it should look like):
[image]

Notice that the blue bar graphs and the "Current User" and "Current Project" text are not showing. I thought at first the data elements just needed time to load, but after giving it multiple hours, it's clear they will not.

I am using a Raspberry Pi 2 Model B

FullPageOS versions tried: 
2019-12-01_2019-09-26-fullpageos-buster-lite-0.11.0
2020.07-01_2020-05-27-fullpageos-buster-lite-armhf-0.12.0

guysoft, 2020-07-20:
What browser you are using where it does work?
It looks from my browser as intended - its blocking a CORS call
The browser also gives me on the console:
```
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://sentry.adafruit.us/api/6/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=8dc5a2741f68419ca4148c629a41be39. (Reason: CORS request did not succeed).
```

It does the same behaviour on my Firefox on my laptop, also adding the network output which says there is a security error
[image]
:

MODICA-Will, 2020-07-20:
Thanks for your quick reply and for trying it out yourself. Admittedly, the other browsers we've been using are all on Windows, so there may be a difference in request handling there. Seems FullPageOS and the Linux browsers are acting in a safer way. 

After a little more testing, we found that if we manually log in to the full Adafruit IO account first, the dashboards display properly. The public dashboard option was supposed to be a way to share viewing access to anyone with a link, without logging in, but it seems that doesn't work properly due to the cross-origin requests. 

Seems at this point that the issue is not with FullPageOS. Thanks again for your help.

guysoft, 2020-07-21:
There are addons to supress cross-origin, but you will have to manually install them. Using crl+t will open a tab you can use.

You wrote "we", so I am not sure if you are for or non profit.  I will note that if FullPageOS is helping you in your business please consider making a donation, since am a volunteer maintaining this 100% out of my free time, link in the readme.