guysoft/FullPageOS #344: Local host webpages

pukpuklouis, 2020-07-27:
FullPageOS has a web server running. I put a HTML file into /var/www/html 
And I put URL into fullpageos.txt 
http://loclahost/HTMLfile.html

My assumption is it can load the local server file without internet connection. But seems like it doesn't work this way ?
Is there any work around can solve this problem?

closed 2020-07-27 by pukpuklouis

guysoft, 2020-07-28:
Hey, did you solve you issue? Can on comment also how you did so others can see?
Thanks

pukpuklouis, 2020-08-05:
When I edit url in fullpageos.txt I try to use comment out. 
#url1
url2

Turns out this txt file is a placeholder for loading url. 
So I just put clean localhost url in fullpageos.txt and it works perfectly.