guysoft/FullPageOS #348: Black Screen on boot

maximilianotaverna, 2020-08-31:
With the latest build on Raspberry Pi4 (4GB) I get a black screen with HDMI plugged into either HDMI0 or HDMI1 . Tried adding a url to fullpageos.txt and still nothing

dresslerc, 2020-08-31:
Whats your screen resolution?  Currently have the same issue with a 4K screen.

maximilianotaverna, 2020-08-31:
Thats a good point @dresslerc , I didn't think of that Im on a 2560x1080/60HZ..

guysoft, 2020-09-01:
What is the "latest build"? Stable or nightly? 
Please provide full version.

dresslerc, 2020-09-01:
2019-12-01_2019-09-26-fullpageos-buster-lite-0.11.0.zip - The current official release is where I get a black screen on a 4K Samsung TV on Raspberry Pi 4.

The nightly 2020-07-01_2020-05-27-fullpageos-buster-lite-armhf-0.12.0.zip works fine on a 4K TV with Raspberry Pi 4.

guysoft, 2020-09-01:
I guess the firmware is update on PiOS which the nightly is based on. So I suggest @maximilianotaverna you use that.

Rezpel, 2021-02-18:
I got the same issue on my Raspberry Pi 4 (8gb) with the version 2020-07-01_2020-05-27-fullpageos-buster-lite-armhf-0.12.0.zip. When booting I see the command line.

kylehase, 2021-02-28:
I'm not sure if this is the same cause but I'm also getting a black screen just after the scrolling boot text sequence.

Ver 2020-07-01_2020-05-27-fullpageos-buster-lite-armhf-0.12.0

Fresh install on RPI 3 with RPI 7" touchpanel with WiFi configured.

Rezpel, 2021-03-03:
@kylehase have you tried apt-get update & apt-get upgrade? That fixed the problem for me.

guysoft, 2021-03-03:
Hey manage to get a report from @amitdar that indeed the nightly has solved an issue. It seems there is a new raspberry pi 4 hardware out there that only works with the nightly. So I intend to build a new nightly with the latest firmware and release that so we don't get those issues.
I am not sure what exactly was changed, and unless someone wants to dig in, just releasing seems to be the solution.

guysoft, 2021-03-03:
There is a new RC now, welcome to test: 
https://github.com/guysoft/FullPageOS/issues/368