guysoft/FullPageOS #352: Why is lighttpd needed?

emil-nasso, 2020-09-15:
#### What were you doing?

Trying out fullpageos.

#### What did you expect to happen?

Didn't expect it to be running a webserver.

#### What happened instead?

It was running a webserver. :)

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

0.11.0

#### Question...

I am trying out fullpage os but I noticed that a lighttpd server was running on the pi. Why is this needed? I uninstalled it but after rebooting my remote website won't show anymore. Does the website get proxied through lighttpd in any way or why is it required?

guysoft, 2020-09-15:
Its needed because it serves the FullPageDashboard service that lets you pin multiple tabs from one webpage. See https://github.com/amitdar/FullPageDashboard
If you are not using it you can safely disable the systemctl service.

closed 2020-09-15 by guysoft

emil-nasso, 2020-09-15:
> Its needed because it serves the FullPageDashboard service that lets you pin multiple tabs from one webpage. See https://github.com/amitdar/FullPageDashboard
> If you are not using it you can safely disable the systemctl service.

Ah. Ok. Is there a way to switch away from it to only display a single website and not have a need for the lighttpd?

guysoft, 2020-09-15:
Sure, its in the documentation. Look at the readme.

emil-nasso, 2020-09-16:
Sorry, I can't seem to find it. :( The readme tells me that FullPageDashboard is the default but I don't can't find anything about switching to just a plain website using chrome.

emil-nasso, 2020-09-16:
Just to clarify:
I put my external website into `fullpageos.txt`, replacing the reference to fullpagedashboard there.
I then uninstalled lighttpd and not my external website doesn't show anymore.

guysoft, 2020-09-16:
You need to add `disabled`` in ``/boot/check_for_httpd`.
See https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/run_onepageos