guysoft/FullPageOS #355: 64it build not avilable yet due to depenecy issues

guysoft, 2020-10-22:
#### What were you doing?

Trying to builds FullPageOS using the new arm64 build variant

#### What did you expect to happen?

For it to build

#### What happened instead?
```
+ apt-get install -y --force-yes chromium-browser
16:08:05 Reading package lists...
16:08:06 Building dependency tree...
16:08:07 Reading state information...
16:08:07 Some packages could not be installed. This may mean that you have
16:08:07 requested an impossible situation or if you are using the unstable
16:08:07 distribution that some required packages have not yet been created
16:08:07 or been moved out of Incoming.
16:08:07 The following information may help to resolve the situation:
16:08:07 
16:08:07 The following packages have unmet dependencies:
16:08:08  chromium-browser:armhf : Depends: libcups2:armhf (>= 1.4.0) but it is not going to be installed
16:08:08                           Depends: libgtk-3-0:armhf (>= 3.9.10) but it is not going to be installed
16:08:08                           Depends: chromium-codecs-ffmpeg-extra:armhf (= 78.0.3904.108-rpt1) but it is not going to be installed or
16:08:08                                    chromium-codecs-ffmpeg:armhf (= 78.0.3904.108-rpt1) but it is not going to be installed
16:08:08                           Depends: libraspberrypi0:armhf but it is not going to be installed
16:08:08                           Recommends: chromium-browser-l10n:armhf but it is not installable
16:08:08 W: --force-yes is deprecated, use one of the options starting with --allow instead.
16:08:08 E: Unable to correct problems, you have held broken packages.
16:08:08 ++++ echo_red 'build failed, unmounting image...'
16:08:08 ++++ echo -e -n '\e[91m'
16:08:08 [91m++++ echo build failed, unmounting image...
16:08:08 build failed, unmounting image...
```

kjake, 2022-04-27:
wouldn't you want to install chromium-browser:arm64 instead of chromium-browser:armhf ?

guysoft, 2022-05-02:
Indeed , does that existing?
will note I am not maintaining this atm so of anyone wants to help they are welcome

kjake, 2022-05-02:
I do see arm64 builds in the repo as of Sept 2021:
http://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/

I can take a stab at updating things - would I just follow the notes in the README?
https://github.com/guysoft/FullPageOS#build-fullpageos-from-within-fullpageos--raspbian--debian--ubuntu

guysoft, 2022-05-10:
I recommend you use the docker build method:
https://github.com/guysoft/CustomPiOS/wiki/Building-with-Docker