guysoft/FullPageOS #356: Is the current build compatible with RPi4 and GPU acceleration?

martinlombana, 2020-10-24:
I am evaluating buying some rPI4s for a project, but I want to make sure that they are compatible with GPU accelation in FullPage OS, because I want to display web animations and videos. I have been looking trought gitHub but I am unsure if this is yet supported or not. If you can confirm, it would be great.

Thanks!

guysoft, 2020-10-24:
Should work, but depends on the gpu component you require

martinlombana, 2020-10-24:
Hummm,
Thanks! What are the ones that do not work? Or what do you mean in
particular?

By the way, I just donated another 10$ 🙂.
-- 

___________________________________

*Martín Lombana*

*TODOLOGIC.COM <http://TODOLOGIC.COM> | APPS & MORE*

*+34 633 63 02 35  |   *+34 946 759 137

*INSTAGRAM *→* @ <https://instagram.com/todologic>TODOLOGIC
<https://instagram.com/todologic>*

guysoft, 2020-10-24:
Thanks!

You can open ``chrome://gpu`` tab and see. I think the 4 and 3B+ should be about the same.
You should get there a list of what is supported - if you say what you are trying to run it might help narrow down. But in the end you just have to plug it and see what happens, because no one really tests and reports this to my knowledge. The GPU is partly closed source.

martinlombana, 2020-10-24:
Cool. I will do some testings then. Thanks!

ilyai, 2022-05-10:
Related issue: https://github.com/guysoft/FullPageOS/issues/29

guysoft, 2022-05-10:
@ilyai Appropriate the help, though that issue has existed well before Rpi4 existed and I am not sure how related it is.

ilyai, 2022-05-11:
@guysoft well, it contains solutions / workarounds for this issue that work on Rpi4 as well