guysoft/FullPageOS #358: WPE Webkit option

martinlombana, 2020-11-05:
Just out of curiosity: Have you tried something with WPE webkit? It seems like it should run faster that the regular chrome. But I am unsure if it works perfectly or not. 

https://wpewebkit.org/

guysoft, 2020-11-05:
Nope, the truth is I barely find the time.
But you should be able to install it, then change the execution command in this file:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/start_chromium_browser

martinlombana, 2020-11-05:
Ah, Interesting! :). Thanks. I will try it then and report when I have a bit of time.

guysoft, 2020-12-20:
any news on this in the end?

darconeous, 2021-03-18:
I'm also curious

waltermvp, 2022-03-19:
I'm curios and willing to help if anyone could point me in the right direction, or provide a few hints :)