guysoft/FullPageOS #360: FullPageDashboard Not Working

mrosu-sp, 2020-12-01:
#### What were you doing?

Trying to setup up FullPageDashboard with two tabs.

I set the URL to http://localhost/fullpagedashboard

#### What did you expect to happen?

The two URLs to start alternating every 30 sec.

#### What happened instead?

404 not found.

I can confirm that this is not a problem with my links.
If I set the links in fullpageos.txt they work just fine.
I also tried a single link on fullpagedashboard like google.com

Also:
http://localhost/fullpagedashboard -> 404 Not Found
http://localhost/FullPageDashboard -> redirects to http://localhost/welcome 

#### Was there an error message displayed? What did it say?

404 Not Found.

#### Version of FullPageOS?

0.11.0

#### Screenshot(s) showing the problem:

[image]

[image]

#### If you are building FullPageOS - provide a build.log that is created for the build

I did not build it.

guysoft, 2020-12-02:
URLs are case sensitive. Its working as intended 
https://www.w3.org/TR/WD-html40-970708/htmlweb.html#h-4.1

guysoft, 2020-12-02:
@tailorvj Can you please fix that link? Its wrong.

Thanks ❤️

guysoft, 2020-12-02:
@mrosu-sp You are seeing FullPageDashboard, there is an iframe that shows the welcome page.
You need to click on this button to the tab, that is FullPageDashboard:
[image]

mrosu-sp, 2020-12-02:
Oh I see.  Now it works mostly as I expected it.  Thank you for this information.

As a test, I was able to add additional tabs and they cycled properly.  The tabs are also persistent through reboot and I am able to delete the welcome page tab.

However, I would like to configure this in the terminal, without the GUI.

1. Can we disable the iframe with the welcome page and pull the links directly from fullpagedashboard.txt?  The instructions make it seem that all we have to do is add links in fullpagedashboard.txt and everything is pulled in that way automatically.

2. Can we remove the header?
[image]

guysoft, 2020-12-02:
0. FullPageDashboard will load the urls from fullpagedashblard.txt on first boot. If you set it using the commadline before it will work. Otherwise you will need to add code yourself 
1. You need something to cycle though links that thing is FullPage dashboard.  If you disable it you loose the functionality.
2. Nightly 0.12.0 has the option to hide the header.

mrosu-sp, 2020-12-02:
0.  It's not working for me =(  It's just not using the links from fullpagedashboard.txt at all.
1.  I see.   In that case I have to manually remove it after first boot to remove it completely.  I understand.
2.  Cool thanks.

guysoft, 2020-12-02:
0. you need to reslash the image, I am stressing again - it only works ONCE.
1. FullPageDashboard has been abandoned by the original developer, you are welcome to add the features you need to it and they will be merged if they dont break anything.

closed 2020-12-02 by guysoft

mrosu-sp, 2020-12-03:
Sorry for the misunderstanding.  I didn't catch that.

I will attempt to work with the nightly versions and see if that works better for me.

I wish I could code ;)

guysoft, 2020-12-03:
You might be able to pay someone on fiverr too, or you could learn and fulfill your wish ;)