guysoft/FullPageOS #364: Raspberry Pi 4 - configuration dual screen - switching ( mirror / extended )

U-Dope, 2021-01-06:
Hello, thank you very much for this Pi development.
With my club, I've been using FullPageOS for at least 4 years on sports events to display results.
At the beginning, I was using Raspberry Pi2 then Pi3 and for this year, I'd like Pi4.

The power of Pi4 could allow me to manage 2 HTML pages simultaneously on 2 differents displays but...
I have been faced with several problems that I may or may not have been able to overcome.

#### What were you doing?
Trying to configure Pi 4 with 2 HDMI output (*)

#### What did you expect to happen?
Trying to display 2 distinct pages on 2 distinct screens
In fact I want to obtain exactly what «[McLund](https://github.com/McLund)» doesn't want in this post : [#302](https://github.com/guysoft/FullPageOS/issues/302)

#### What happened instead?
The 2 chromium instances launched, displayed one on each other because of mirror screen mode.

#### Was there an error message displayed? What did it say?
Not at all, everything seems to work perfectly (at least, as I don't want it to)

#### Version of FullPageOS?
I've tried with this image "2020-05-13_2020-02-13-fullpageos-buster-lite-0.12.0".
But this one "2019-12-01_2019-09-26-fullpageos-no-acceleration-buster-lite-0.11.0" seems to have the same behaviour.

(*)
- I didn't edit anything to /boot/config.txt file
- I have modified /etc/dhcpcd.conf
DHCP is great but... but I need Static IPv4 on my network.
Whereas previous versions of Raspbian allowed easy configuration of an static IP on a wired network in the file [boot/fullpageos-network.txt]. New versions is a real pain for this: fullpageos-network.txt disappeared and a new file appeared [boot/fullpageos-wpa-supplicant.txt] (which seems useless for wired network)
Connexion via SSH is required to edit /etc/dhcpcd.conf (with configuration lines similar to file [boot/fullpageos-network.txt])
(In fact, I hope I'm bullshitting on this point of cable network configuration...)
- I have modfied /home/pi/scripts/start_chromium_browser file in adding these 2 lines :
```
chromium-browser --new-window --user-data-dir=/tmp/browser-1 --window-position=   "0,0" --start-fullscreen --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --simulate-outdated-no-au="Tue, 31 Dec 2099 23:59:59 GMT" --disable-component-update --overscroll-history-navigation=0 --disable-translate --disable-infobars --disable-features=TranslateUI --disk-cache-dir=/dev/null --app="http://192.168.1.0/page1.html" &
chromium-browser --new-window --user-data-dir=/tmp/browser-2 --window-position="1920,0" --start-fullscreen --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --simulate-outdated-no-au="Tue, 31 Dec 2099 23:59:59 GMT" --disable-component-update --overscroll-history-navigation=0 --disable-translate --disable-infobars --disable-features=TranslateUI --disk-cache-dir=/dev/null --app="http://192.168.1.0/page2.html"
```

- I have tried (without success) to modify /etc/lightdm/lightdm.conf file in that way :
`display-setup-script=xrandr --output HMDI-1 --left-of HMDI-2`
(but xrand command seems inoperant)

- I read somewhere that I should disable 3D to access the extended display mode.
Well, I'm a bit lost... If someone could lead me in the right direction, that would be great.
Thanks

guysoft, 2021-02-07:
Not sure how to do this, dual screen is the only documented #302

U-Dope, 2021-02-10:
Finally, I gave up with configuration of FullOS and Pi4...
I used image « 2020-12-02-raspios-buster-armhf-full.img » include in « raspbian_latest.zip » from [raspberrypi.org](https://downloads.raspberrypi.org/raspios_full_armhf/images/)
This image OS natively boots with the extended screen on dual screen.
Then I configured :
1 - sudo nano /etc/xdg/autostart/myapp.desktop
```
[Desktop Entry]
Exec=lxterminal -e /home/pi/bin/launch_chrome.sh
```
2 - sudo nano /home/pi/bin/launch_chrome.sh
```
chromium-browser --new-window --user-data-dir=/tmp/browser-1 --window-position=   "0,0" --start-fullscreen --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --simulate-outdated-no-au="Tue, 31 Dec 2099 23:59:59 GMT" --disable-component-update --overscroll-history-navigation=0 --disable-translate --disable-infobars --disable-features=TranslateUI --disk-cache-dir=/dev/null --app="http://192.168.1.0/page1.html" &
chromium-browser --new-window --user-data-dir=/tmp/browser-2 --window-position="1920,0" --start-fullscreen --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --disable-session-crashed-bubble --simulate-outdated-no-au="Tue, 31 Dec 2099 23:59:59 GMT" --disable-component-update --overscroll-history-navigation=0 --disable-translate --disable-infobars --disable-features=TranslateUI --disk-cache-dir=/dev/null --app="http://192.168.1.0/page2.html"
```
3 - [Screen Configuration](https://www.raspberrypi.org/blog/buster-the-new-version-of-raspbian/)
« _GL driver is in use, there’s a new ‘Screen Configuration’ tool – this enables you to set up the arrangement of multiple monitors on a Raspberry Pi 4_ »
4 - Disable screen blanking
« _The file /boot/cmdline.txt contains a single line of text.
Add consoleblank=n to have the console blank after n seconds of inactivity.
For example consoleblank=300 will cause the console to blank after 300 seconds, 5 minutes, of inactivity.
Make sure that you add your consoleblank option to the single line of text already in the cmdline.txt file.
To disable screen blanking, set consoleblank=0.
You can also use the raspi-config tool to disable screen blanking.
Note that the screen blanking setting in raspi-config also controls screen blanking when the graphical desktop is running._»

guysoft, 2021-02-11:
Thanks for the update. Not sure how integrate that in to FullPageOS, if anyone comes here and does input would be appreciated :)

guysoft, 2023-10-25:
Related #302